d-ronin/dRonin #829: GCS crashes on startup in linux (next branch)

thunderdan, 2016-03-23:
I've seen this for a couple of days on my machine.  The gcs won't start and the error message is below.

ASSERT failure in QVector<T>::at: "index out of range", file /home/dmorgan/code/dronin-readonly/tools/Qt5.5.1/5.5/gcc_64/include/QtCore/qvector.h, line 393

I wouldn't think anything is wrong with the qvector.h file providied w/ Qt so probably something in the usage of it is wrong.

A dump file is written to disk so i can upload if that helps.  I could also back track to see when the issue was introduced if you like.

thunderdan, 2016-03-23:
I tried earlier commits and both crash on startup for me

mlyle, 2016-03-23:
Try resetting the ground configuration -- maybe you've gotten that into an invalid state -- `./build/ground/gcs/bin/drgcs -r`

If this doesn't work, can you try running from this binary distribution?  http://jar.lyle.org:8080/job/dronin/263/Nodes=master/artifact/build/artifacts/dronin-20160322-2f65e4b7.tar.xz

tracernz, 2016-03-23:
Running with the debugger from Qt creator will show you exactly where this is coming from.

thunderdan, 2016-03-24:
Before I tried your other suggestions, I did `make gcs_clean` followed by `make gcs`.  The problem goes away and  I can run the GUI again now.  

As a side note: Running dronin from the binary distro from January has always been fine.  The about dialog displays "Release-20160120.3"

Do the settings for gcs built from src get stored in a separate location than the settings when installed from binary?

I'll close the issue and if I see any issues like this again I'll try to get more info with the debugger.

closed 2016-03-24 by thunderdan

mlyle, 2016-03-24:
No, but there are known issues with dependencies in the build that can result in a crashy GCS from incremental builds of wildly different versions.