d-ronin/dRonin #830: Autotune produces extremely large values with undermount frame

reubenfirmin, 2016-03-24:
I attempted autotune on a new build (BLHeli 12A ESC, 2204/2300 motors, hobbyking undermount FPV250LH frame). Center of gravity is significantly lower than the QAV250s that I've successfully autotuned in the past, since the battery is on the bottom. Even after 3 attempts (fairly stable hovers without wind) it still recommends values that cause warnings on the stabilization page. Does the algorithm not cope with unusual configs?

mluessi, 2016-03-24:
Have you tried flying it? It could be that the warning values are conservative and the gains actually work for your frame.

pug398, 2016-03-24:
Is it the warning about tau being high (which is ok) or on the Basic tab you get values in the red? P term in red probably would not be safe to fly but other values high can be caused by lower center of gravity.

dustin, 2016-03-24:
Consider sharing and linking to your tune as well. We can learn a lot from
looking at those.

droidicus, 2016-03-25:
I will take a look at this when more information has been provided.

reubenfirmin, 2016-03-26:
@droidicus @dustin : https://dronin-autotown.appspot.com/at/tune/ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICAgNi2owkM

@pug398 : 
[image]

mlyle, 2016-03-26:
That's what happens when no autotune flight has been flown or the autotune data was not saved.

To fly autotune, enable an autotune flight mode and the autotune module.  Take off with PIDs set to default, and select the autotune flight mode.  The quad will wiggle for 60s, then land and disarm, **while staying in the autotune flight mode** to save values.

Then you can do the autotune process like you have.

closed 2016-03-26 by mlyle

reubenfirmin, 2016-03-26:
@mlyle That's exactly what I did. I've autotuned other quads successfully a dozen times. So why isn't this saving?

reubenfirmin, 2016-03-26:
@mlyle Please remove the "invalid" label. This is definitely a bug.

mlyle, 2016-03-26:
I don't know what you did differently.  Are you sure you disarmed while still in the autotune flight mode?

reubenfirmin, 2016-03-26:
@mlyle I am sure. I've done this three times over now.

mlyle, 2016-03-26:
You have changed and done something differently, or you are using mismatched GCS and firmware.  I'm sorry.  There have been thousands of autotunes.  Some have invalid data from capturing system state wrong, but none have a failure to save the system state.

People definitely tend to switch to leveling and not save-- we have fixed that in next.

reubenfirmin, 2016-03-26:
I definitely didn't switch to leveling, I promise you.

mlyle, 2016-03-26:
Does your GCS match the firmware you're running?

mlyle, 2016-03-26:
I guess there's one other unlikely thing it could be, too.  There are other bugs for the settings filling up with objects from old releases, etc.  If you've flown this on Tau and OP and LP your settings partition could be full and it could be unable to save the autotune data.  

You can export the UAV configuration in GCS, then erase the settings partition (what flight board?  There's no great way to do this on CC3D unfortunately), and then import it.

pug398, 2016-03-26:
I missed something. It looks like a failed tune to me due to arm flex and can be checked by using lighter battery and/or strapping dowels across diagonal arm pairs to reinforce. Vtails will fail autotune regularly with out of range P term for this reason.

mlyle, 2016-03-26:
@pug398 -- the tune object is all 0's, 0 observations.  This only happens when it fails to save or the GCS can't read the tune object.

mlyle, 2016-03-26:
@reubenfirmin -- you could try going to latest next, which as part of upgrade will automatically upgrade the settings partition and clear out old objects, too (the code to erase objects not applicable to the release was not in OP or Tau).   e.g. http://jar.lyle.org:8080/job/dronin/lastSuccessfulBuild/Nodes=winx86/

pug398, 2016-03-26:
Got it now and see it on autotown. I knew there must have been a reason for the line of thought just didn't make the connection. I have been able to link consistent AT failures back to frame design or too heavy a payload. Hopefully the new release will allow a good tune to be saved.

mlyle, 2016-03-26:
@pug398 just to clarify-- frame design / too heavy a payload results in like, 4ms tau or bogus betas, right?  Not a complete failure to save any data (I don't see how that could happen / have not seen that before).

pug398, 2016-03-27:
That is correct it completes and you can save the data. Just generates extremely high P terms that are hazardous to the quads health. I didn't follow his link to autotown or I would have known what you were referring to. I actually have seen tunes on vtails and flame wheel clone frames that seemed so similar I jumped to conclusions. Not the first time as you know lol. Those frames with heavy payloads drive autotune crazy.

fujin, 2016-03-29:
Confirmed frame with significant Z-offset lower CG (110g battery mounted below lower-bottom plate, no HD camera, dry weight = 308g) with 1kHz/188lpf/60Hz gyrocutoff @ next was able to successfully tune @ 10ms. Slightly higher than the aeroelastic limits. Tune flies perfectly well even with considerable positive Z-offset (adding HD camera). Would advise investigating mechanical components health, ESC configuration/calibration and potentially flight controller vibration isolation via o-rings.

mlyle, 2016-03-29:
@fujin-- his tune was never saving.  He either had a full config partition or was running hte procedure wrong.  His systemident object in the shared tune is all defaults.

reubenfirmin, 2016-03-31:
I'm going to do another test at the weekend after flashing the board in rescue mode. Stay tuned...

mlyle, 2016-03-31:
If it's CC3D, note that that does not erase the settings flash.  You may want to try the next build at https://github.com/d-ronin/dRonin/issues/861 which now uploads a special program to CC3D to retrieve and clear settings flash as part of upgrade.

The issue is this-- there is no mechanism for garbage collecting configuration objects that are no longer relevant.  Every release you have run might have different versions of ManualControlSettings and other objects--- and once the maximal number of objects are stored it's not possible to save any new ones.

Since new-upgrade now erases settings flash as part of upgrade, this isn't a concern anymore.

reubenfirmin, 2016-03-31:
Got it. OK, I will try the new build.

reubenfirmin, 2016-04-05:
Well, unfortunately for the forensics here, RIP underslung 3d-printed frame. It took a bad crash at the weekend and is no longer air worthy :/