d-ronin/dRonin #831: Selecting Oneshot in VSW with DALRC BLHeli's fails to spin motors on testing step

reubenfirmin, 2016-03-24:
I have Crazepony DALRC 12A ESC Oneshot125 ESCs on a new build. 

Couple of issues:
1) I had to synchronize ESCs manually before the vehicle setup wizard was able to spin them up. Perhaps the vehicle setup wizard should have a "synchronize ESCs" option / step?

2) If I selected "oneshot" in the VSW, the motors wouldn't spin up. Is the Oneshot on these ESCs a different protocol than the one Dronin supports?

mluessi, 2016-03-24:
What do you mean by "synchronize ESC", do you mean calibrating them?

reubenfirmin, 2016-03-24:
You are correct, sorry. Filed this without referring to the UI, low on coffee.

mlyle, 2016-03-24:
@reubenfirmin -- what flight board are you using and what kind of RC input?

CC3D doesn't support oneshot at the same time that PWM-in (one wire per channel) is used because of how the timer resources are allocated.

mlyle, 2016-03-26:
Can you answer the question here please?

closed 2016-03-27 by mlyle

reubenfirmin, 2016-03-31:
@mlyle (You should give bug reporters longer to respond to tickets before closing them. I've found when running open source projects that 2 weeks is about the right time before assuming all is lost. Certainly no need to follow up just 2 days after asking a question.)

You are correct - this is a CC3D. I suggest that if dronin detects a board that doesn't support oneshot that it doesn't offer oneshot in the vehicle setup wizard.

mlyle, 2016-03-31:
You were answering other questions at the same time ;P

CC3D supports oneshot; CC3D does not support simultaneous PWM input and oneshot output.   ( #847 )

reubenfirmin, 2016-03-31:
Got it, thanks.