d-ronin/dRonin #852: KISS24RE Actuator Feedback / ESC Telemetry

fujin, 2016-03-28:
It's possible to gather information from the KISS24 Racing Edition ESC such as: voltage, current, RPMs and temparature via serial interface. The protocol is available, as well as an arduino protocol example.

- Temperature (resolution 1deg C)
- Voltage (resolution 0.001 V)
- Current (resolution 0.01A)
- Consumption (resolution 1mAh)
- Electrical RPM (resolution 100Rpm)

The data structure and crc functions are all contained in the attached PDF toward the end of the issue. The only thing you would have to configure is the pole count to get the real RPM.

There's a wiring section regarding a 1k pull-up "one time" which I'm not quite sure how to interpret, if someone could help me figure out how to do this.. is it possible with just a regular 1k resistor + Vcc?

[image]

example mwosd/kissfc vid: https://youtu.be/pup_af6Bu_0

I expect we would want a module to manage some of the tasks, as well as some UAVos? Shared perhaps, or instances(?) I think @mlyle was against instances. I don't really know what to do here so would appreciate some guidance. I know that there is a timing-critical component which has to be done at the actuator level for us, AFAICT

I'd probably just log these uavos at first but the end goal would be to supply the same information (and more) as the Battery Monitoring module without a csen/vdiv when using K24's. I bet we could send it through the MSP bridge as well. 

Maybe even use the RPM data for some automated step-impulse response measurement if the equipment was able to be fixed to the ground and adequately secured, safety-wise.

[KISS_telemetry_protocol.pdf](https://github.com/d-ronin/dRonin/files/192496/KISS_telemetry_protocol.pdf)
[KISS24ARE_Telemetrie_arduino_example.zip](https://github.com/d-ronin/dRonin/files/192497/KISS24ARE_Telemetrie_arduino_example.zip)

mlyle, 2016-03-28:
I was talking to @ekaszubski some about this last night.

The most important / first thing is to change the alignment of the PWM and oneshot we make-- to not have aligned leading edges, but aligned trailing edges instead.  This will A) remove phase-modulation of output motor power, and B) make all of the channels "free up" at the same time.

This requires plumbing the true channel maximum values through into PIOS_Servo and using them to choose rise times of the pulse, and have everyone have the pulse fall just before cycle completion / timer stop.

Next, what's required is a mode for when all the PWM is completed, to set the signal to a telemetry-mode specific state (e.g. leave it alone if there's no telemetry, set it high-z for most other formats, etc).

Finally, the logic that re-enables the timer needs to be adjusted to see whether that telemetry mode is "done" and whether the update can be processed.

@ekaszubski wants to do castle link telemetry, which involves receiving (effectively) PPM on the pin afterwards.  And this has similar needs.

fujin, 2016-03-28:
@mlyle the 30uS can supposedly be sent independently of the normal throttle signal and independent of the signalling speed as well, not sure if that changes any of the above or if I'm just misinterpreting :book:

mluessi, 2016-03-28:
The KISS ESCs are amazingly efficient, check out the current values in the video @fujin posted ;)

closed 2017-07-10 by mlyle