d-ronin/dRonin #861: Tanto Beta Issue

mlyle, 2016-03-31:
For keeping track of rev, non-bug feedback, relnote draft, etc.

Githash: 4c53eeeb47601920e2e0614f94c25b640f7675ca

Artifacts:

* Windows: http://jar.lyle.org:8080/job/dronin/281/Nodes=winx86/
* OSX: http://jar.lyle.org:8080/job/dronin/281/Nodes=osx/
* Linux/Android: http://jar.lyle.org:8080/job/dronin/281/Nodes=master/

fujin, 2016-03-31:
woo!

cpusmith, 2016-04-01:
[image]

Should the selection boxes be so wide?

mlyle, 2016-04-01:
@cpusmith -- change @tracernz made, I thought it was kinda cute.

mlyle, 2016-04-01:
Relnote rough draft:

Top two of these should link to wiki pages

* Upgrade simplified for all targets except Naze32 and PipXtreme.  To upgrade, just start the ground control system and then plug in the flight board, and carefully follow the on-screen prompts.
* HangTime added, which allows controlled maneuvering for a limited time at zero throttle in multirotors.  HangTime should be used with switch arming and power settings that are well under hover power.
* Intelligent clipping management added to improve vehicle performance when motors are near minimum or maximum speed.  In these cases, an "Actuators" alarm will fire that is visible on on-screen displays (Brain, MWOSD).
* The Naze32 rev6 flight controller is now supported.
* Telemetry and configuration subsystem is now considerably faster to connect and more reliable.
* Many modules are now automatically enabled or disabled based on port configuration.
* Further improvements to CCPM helicopter flight modes and configurability.
* Support for the Storm32 Brushless Gimbal Controller across most targets.
* Support for sending GPS data across MSP protocol to MWOSD.
* Support for OpenLog across most larger targets.
* Support for non-inverted SBus, for ports that do not have an integrated inverter or for radios that send non-inverted SBus.
* Significant improvements to performance of altitude-hold mode.
* Additional port types and configurations are supported on the Revolution, Quanton, and Colibri flight controllers.
* External magnetometers supported on the Revolution flight controller.
* New failsafe flight mode to allow deliberate triggering of failsafe behavior with flight mode switch.
* Improvement to board hardware configuration for Lumenier LUX and simulator targets.
* Retrofit of USB subsystem in ground control software eliminates a major cause of crashes and hangs.
* Included minimal version of Android GCS (as a .apk file for side-loading).
* Vehicle setup wizard now follows best practices-- no standard PWM output, enables spin while armed behavior.
* Fred resources-- CPU, memory, and flash-- across all flight boards
* Board-specific ADC pin names display in GCS to improve battery module usability.
* Ensured consistent support for receiver and telemetry protocols across ports on all targets.
* Renamed targets for usability: "revomini" is now "revolution", "coptercontrol" is now CC3D.
* Acro+ insanity factor is now configurable on the expert stabilization configuration pane.
* Actuator curve fit default value changed to 0.9; it is now configurable on the expert stabilization configuration pane.
* "Apply" button removed from most screens of GCS in favor of a single "Save" button.
* Significantly iproved ialog to import saved UAV configurations.
* Bugs where objects could fail to save in the UAV Object Browser were corrected.
* Input wizard is much less likely to falsely detect axis activity.
* Various other usability and look and feel improvements to the ground control software.
* MWRate and Throttle PID Attenuation are now deprecated and will be removed in a future release.
* Development infrastructure and upstream libraries upgraded: Arm 2015Q4 tools, qt-5.51, STM32 Libraries.
* Ease of configuring Windows development environment improved.
* Python API augmented to have additional functionality to read configurations and to more flexibly handle logs.
* Windows installer now ensures that old files from previous installations are removed to prevent erratic behavior.

fujin, 2016-04-01:
typo:
> Fred resources-- CPU, memory, and flash-- across all flight boards

fujin, 2016-04-01:
`CCPM` with regard to helis is? collective.. pitch..? not CPPM right?

fujin, 2016-04-01:
omg sorry pressed close by acccident

fujin, 2016-04-01:
typo:
> Significantly iproved ialog to import saved UAV configurations.

are you going to proof/spellcheck this yet, @mlyle ?

mlyle, 2016-04-01:
I'm going to work on editing it more, yes

mlyle, 2016-04-01:
I think however I cut and paste it ate some letters :P

tracernz, 2016-04-01:
@mlyle The change wasn't really intentional, just a side effect of not explicitly setting the widget size. :-P

fujin, 2016-04-01:
@mlyle happy to read over before it's published :) looks like some simple stuff that gspell or whatever would pick up.

pug398, 2016-04-01:
In a settings file I know the order of the objects does not matter but how about the order of the fields within a object? Trying to track down an issue with importing pathplanner settings.

mlyle, 2016-04-01:
It shouldn't, no.. 

I haven't tried to repro the pathplanner thing yet, but I've audited a lot of code.  The flight side there does some slightly odd things (not new) but they don't look wrong, per se.

mlyle, 2016-04-01:
Relnotes mostly done and ready for any review.

mluessi, 2016-04-01:
Looks good to me :)

tracernz, 2016-04-03:
Rel notes looking pretty good now. Maybe /s/now CC3D/now "cc3d"/ to be consistent.

@fujin Missed your comment earlier, https://en.wikipedia.org/wiki/Cyclic/collective_pitch_mixing ;-)

pug398, 2016-04-03:
Is android GCS supplied in Master suppose to be a copy of the dev signed apk or exported without any key?

mlyle, 2016-04-03:
@pug398 -- for the GCS that will ship with Tanto, it'll be an unsigned image for side-loading.

#705 is the bug where we brought the build infrastructure back, and deferred a couple of remaining issues (maps API, store, signing) to another release.

(We've not enregistered in the google play store or the Maps API program etc yet).

pug398, 2016-04-03:
@mlyle So to date the unsigned apks are just placeholders and are not actually suppose to install on a device?

mlyle, 2016-04-03:
@pug398 -- I've installed them on my device.. I just have sideloaded it with bluetooth and started it.

mlyle, 2016-04-03:
(Am I missing something?  I don't think I did anything special to use our artifact)

pug398, 2016-04-03:
I don't know. None of them will install for me and I have "non Play" security turned off.

mlyle, 2016-04-03:
What android version?  @tracernz what have you used-- you've been successful right?

tracernz, 2016-04-03:
Perhaps you need to have your device in developer mode to install it? Settings > About phone and tap "Build number" seven times. I loaded it using `make androidsdk_install` which uses ADB, so debugging needs to be enabled for that method.

mlyle, 2016-04-03:
Hm, I am not in developer mode; on moto x pure on android 6.

pug398, 2016-04-03:
@tracernz 
Thank you for that tidbit of info. That is way cool. Unfortunately apk still will not install.

Android version 4.4.2

mlyle, 2016-04-03:
I think I see.  It looks like the apk doesn't work in the release configuration.  Fixing.

pug398, 2016-04-03:
Before I forget again there are a couple of minor issues with browser object descriptions. Not sure if you want to fix them up before release.
Settings>CameraStabSettings - remove one of the m from module
Settings>OnScreenDisplayPageSettings -  add c to all mention of sreen 
Data Objects>FlightTelemetryStats - change openpilot to autopilot maybe
Data Objects>SystemAlarms - change openpilot to autopilot maybe

pug398, 2016-04-03:
Same for Settings>OnScreenDisplaySettings as it looks like it has it's own description.

tracernz, 2016-04-03:
@pug398 I have some of this done in a branch I started working on the other day (it does a lot of other stuff to UAVOs as well). https://github.com/tracernz/dronin/tree/mrc-uavo-stuffs

mlyle, 2016-04-03:
ok, stopping doing it then :)

pug398, 2016-04-03:
Cool. I have been meaning to mention it for awhile but it kept slipping my mind. Doesn't take much you know :)

tracernz, 2016-04-03:
Maybe fix the typos etc. up for this release and I will rebase it out of my branch since it won't land for a while.

mlyle, 2016-04-03:
New beta artifacts posted above

pug398, 2016-04-03:
Android GCS working now. Good job :)

jhitesma, 2016-04-08:
Tried the android GCS last night now that I have a new phone with a connector that isn't worn out.  Seemed to work just fine.  At least as well as the android GCS ever worked.

closed 2016-04-09 by mlyle