d-ronin/dRonin #863: Attitude estimates often poor in fast-forward-flight or in fixed wing

mlyle, 2016-04-01:
Shows up really blatantly on OSD, etc.

taulabs/taulabs#2021 etc

mlyle, 2017-02-02:
This is the real issue with multirotors:

[image]

(From a simulation run, where attitude was altered to gate accelerometer data for 300ms after any fast gyro moves).  The paradoxical movement due to initial acceleration on the x/y accels biases the attitude estimate before steady state is reached.  The 300ms gate improves things a lot, but A) doesn't solve it entirely, and B) probably wouldn't play nice with autotune.

dustin, 2017-06-20:
I've been struggling with this on fixed wing lately.  Roll is OK.  I'm able to fly fine with roll in auto-leveling.  Pitch doesn't work well at all.  With default accel gains, I get something that looks like this (ST1 is leveling on roll, rate on pitch):

[image]

Which leads to a position hold attempt that looks like this:

[image]

Motor's up, nose down -- which it appears to believe is the correct thing.

I reduced accelkp down to 10% its original value and it seemed mostly just kind of messier:

[image]

I can't say for certain that it was net worse, but it didn't seem better.