d-ronin/dRonin #864: Confusing vertical velocity reporting

dustin, 2016-04-01:
I originally reported this in TauLabs/TauLabs#2026 where I took off and immediately had a negative climb rate although the altimeter seemed to be tracking fine.

http://gfycat.com/GlassImpartialHydra

A more recent clip shows that it at least seems to be lagging behind.

http://gfycat.com/WelcomeWeakBlacklab

mlyle, 2016-04-02:
@dustin -- can you try setting AttitudeSettings.VertPositionTau from 2.0 to 0.9?  That's a reduction of 2.2x; it'll make the vertical position filter listen much more to the baro and much less to the Z accelerometer and past trend.

I'm curious if this A) looks significantly better on the OSD, and B) if altitude hold flies all right / better after this.

mlyle, 2018-03-13:
Idea: don't pass DC on accelerometer on cfvert.