d-ronin/dRonin #869: Bootloader: Revision management is bad.

mlyle, 2016-04-02:
1. Revision should be volatile so it doesn't get const-optimized.
2. Revision should default to 0xffff and be short aligned (e.g. be programmable at runtime).
3. BU should preserve revision when doing a boot-update. (instead of failing if revision mismatch).
4. Procedure should exist for programming board revision at commissioning-time.

tracernz, 2016-04-02:
Where possible we should detect revision at runtime (e.g. like we do on Naze32). Need to study schematics of other boards to see where else this might be possible.

mlyle, 2016-04-02:
Yah, 0xffff can mean 'detect at runtime' for any targets that support it.

tracernz, 2016-04-02:
Indeed. :-)

mlyle, 2017-03-15:
I think we should just leave this as a bit of yucky legacy.  Really verifying this to be correct is too hard.

closed 2017-03-15 by mlyle