d-ronin/dRonin #879: Modules Call UAVO Functions Without Checking UAVO Init Suceeded

tracernz, 2016-04-03:
e.g. https://github.com/d-ronin/dRonin/blob/next/flight/Modules/Attitude/attitude.c#L233

One more reason why we crash rather than failing with an alarm when heap fills up.

jpbarraca, 2016-04-19:
Basically, modules will proceed normally without ever checking the initialisation of their subcomponents. I started adding checks to modules, returning an error if an initialisation failed. Should the module also trigger an alarm? I expect that an OUTOFMEMORY alarm will be triggered when the heap is depleted.

mlyle, 2016-04-19:
Better to just let the outofmemory thing handle it, I think.

tracernz, 2016-04-19:
Yep, heap alarm will take care of it. Sounds like you're on the right track.

jpbarraca, 2016-04-19:
The advantage of triggering an Alarm from the module is that besides the heap alarm we would rapidly know which module triggered the problem. But I may be missing something.

tracernz, 2016-04-19:
The thing is, it's really the combination of modules that causes a heap overflow rather than just the last one that attempted init before overflow. I can see that it would allow you to work backwards disabling modules until it stops coming up. I'm not sure that's so useful though.

I wish it were possible to give a little colored bar on the modules tab in gcs showing estimated memory usage but I don't see an obvious way to do that without creating a large maintenance liability.

jpbarraca, 2016-04-19:
@tracernz It would help debug. But I admit that even in this case, debugging will be limited because several modules will trigger alarms simultaneously and there is no real culprit. I will just return a failure and prevent the module from operating.

That bar would be great, but I agree that it would add much complexity.

commit 223a3fce (2016-05-04): flight: Validated UAVO initialization in Modules. Fixes #879 (#932)
closed 2016-05-04 by mlyle