d-ronin/dRonin #883: Duplicate autotune values - multiple autotunes

yamahabest, 2016-04-03:
It seems that I am getting duplicate autotune values, when running multiple autotunes

Steps to reproduce (I believe):
1) Flight mode: Leveling
2) Connect battery
3) Arm
4) Fly
5) Switch the flight mode to Autotune
6) Wait till completion
7) Land
8) Disarm
9) Connect to GCS, watch values, and upload
10) Disconnect from GCS and from USB
11) Disconnect and connect battery (all whilst still having the flight mode set to Autotune)
12) Switch the flight mode to Leveling
12) Arm
13) Fly
14) Switch the flight mode to Autotune
15) Wait till completion
16) Land
17) Disarm
18) Connect to GCS, watch values, and upload
19) All values are exactly the same

It seems that if I switch step 11 and 12, it works correctly.

mlyle, 2016-04-03:
No repro.

mlyle, 2016-04-03:
Possible duplicate #747

closed 2016-04-03 by mlyle