d-ronin/dRonin #885: Use single wire mode for HOTT telemetry

mluessi, 2016-04-03:
HOTT is a single wire protocol, so on F3 and F4 we should use single wire mode for the UART. This makes it easier for users, as the cable only has to be connected to the Tx pin.

@tracernz I think you are using this system, WDYT?

droidicus, 2016-04-04:
Could this also be applied to FrSky S.BUS telemetry?

mlyle, 2016-04-04:
frsky smartport telemetry?  it already is, on F3 (where we have hardware inversion).  it will be on brainre1 as a special case, too.

droidicus, 2016-04-04:
Yea, S.Port, I always get the two confused...

mlyle, 2016-10-07:
Deferred pending any further user complaint on this topic.

tracernz, 2016-10-07:
I wonder if we actually have any HoTT telemetry users. I wonder if I can get that from usage data. (For the record/OP, I don't have any Graupner gear, just Jeti and Spektrum)

mlyle, 2016-10-07:
It's somewhat obscure, and definitely isn't very usable on our setup (both because of this issue and the UI etc).

tracernz, 2016-10-07:
I want to find somebody with Graupner gear who is willing to test some stuff so I can refactor it into the SRXL driver. :wink: We can also test this with the same person if they appear.

mlyle, 2016-10-07:
I'll keep an eye out.

[image]

closed 2017-01-30 by mlyle