d-ronin/dRonin #899: Documentation:  Need to document how to get dRonin-MWOSD

mlyle, 2016-04-06:
And/or make a distribution etc

cc @dustin

tracernz, 2016-04-07:
Same as MWOSD procedure for other FCs except uncomment define DRONIN instead of define X.

mlyle, 2016-04-07:
Yah.  sucks to make people compile stuff etc though.

tracernz, 2016-04-07:
It's the arduino way

mlyle, 2016-05-29:
oh shit, wrong issue number in #1016.

fujin, 2016-06-13:
hmmm.. I don't think this is valid? you just `#define DRONIN` and you are good to go.. the rest is all user-config

mlyle, 2016-06-13:
People have showed up many times and asked about it (and expected preloaded images to work, etc)-- seems like a couple paragraphs in the docs would help.

tracernz, 2016-06-18:
We could pull some of the work in ShikOfTheRa/scarab-osd#222 into a PR and see if we can get Makefile building happening in upstream scarab-osd. That should get us closer.

tracernz, 2016-07-09:
Some brief docs for this release: https://github.com/d-ronin/dRonin/wiki/User-Guide:-Setting-Up-MWOSD. Would appreciate test and feedback.

emaugain, 2016-07-15:
@tracernz I have a micro minimosd so I can test this out.

tracernz, 2016-07-15:
@emaugain I suspect a few of the details could be documented more clearly/explicitly. I'd be greatful for your feedback. :+1:

emaugain, 2016-07-15:
jenkins, build artifacts please

mlyle, 2016-07-22:
https://dronin.readme.io/docs/configuring-mwosd-for-dronin

closed 2016-07-22 by mlyle

emaugain, 2016-07-25:
@mlyle tested this and the guide was clear - i don't see flight mode and armed/un-armed
(used to show on cleanflight) am i missing anything?

dustin, 2016-07-25:
Arm state is implicit.  Detailed flight modes currently requires lots of
extension (as they're also implicit in the base feature set).

It's not clear what you see it what you're doing. There are lots of ways to
get that stuff to not show up in mwosd.

jihlein, 2016-07-25:
I had started a re-work of MWOSD such that it would display armed/unarmed and the Dronin specific modes.  It's on my todo list to finish.

dustin, 2016-07-25:
I've not had an issue with arm/disarm.  Is something broken?

The mode-specific thing is currently done in one of like, three ways in
MWOSD.  :/  The one we're using is kind of implicit based on compatibility
with MW.  It's pretty gross.  There's another that has named modes that go
across the wire, but MWOSD still needs to know all the modes in order to
render them (since it's a font thing).

PR #1016 (2016-05-29): gcs: welcome: improve welcome screen links
Add documentation link; fixes #899
Add forum link.

All links tested, happy.  Still would like some appcasting-like thing.
commit c599ddd0 (2016-05-29): gcs: welcome: improve welcome screen links

Add documentation link; fixes #899
Add forum link.