d-ronin/dRonin #903: STM32F411 Boards

davidbitton, 2016-04-08:
Will the firmware work against an STM32F411 based board? I understand the F405 and F407 use a different library. Thanks!

dustin, 2016-04-08:
Individual board support is going to be work no matter what.  It's likely you can port to new board, but it's not going to accidentally boot up as an existing target.

mluessi, 2016-04-08:
Yeah, we use the latest STM libraries, so a 411 based board could easily be added.

closed 2016-04-08 by mlyle

davidbitton, 2016-04-08:
I'm fine w. the concept. What makes a flight/target just that? I see
there's a discoveryf4 target. Which F4 discovery? Is it the F411? What
about each target makes it unique? If the discoveryf4 is the prototyping
board for the F4, how does one then translate that to a finished product?

mlyle, 2016-04-08:
@davidbitton -- DiscoveryF4 is solely for developers at this point, and doesn't carry a full flight stack-- just the basic telemetry etc subsystems.  We used to have a "FlyingF4" target based on the F407 discovery board, but removed it because it was A) problematic for various reasons, and B) no one used it.

FlyingF3 is a discovery board though-- it's the STM32F3Discovery.

May I ask the purpose of your questions?  Are you developing a flight board?

davidbitton, 2016-04-08:
I am not the developer. I wanted to make use the of the new [Falcon F4](http://tro.pe/1qcE3yL). They do not have current support for dRonin. They only have official support for: 

- LibrePilot 15.09
- RaceFlight Release 1, RC8, Version 16.03.17a
 
 I thought I would try my hand at contributing.

mlyle, 2016-04-08:
@davidbitton -- we'd love help, but.. are you sure you want that board?  There's these shrunk Revolution (without the radio) http://www.aliexpress.com/item/Mini-Openpilot-CC3D-Revolution-Flight-Controller-for-Nano-Mini-RC-Multicopter/32616936566.html?spm=2114.01010208.3.45.xkpZBW&ws_ab_test=searchweb201556_0,searchweb201602_1_10017_10005_10006_10034_10021_507_10022_508_10032_10020_10018_10019,searchweb201603_1&btsid=c44343d4-edf1-44ee-9223-e3dc8b69e7c6 that are about the same price and smaller..

I can't promise we'd accept a new target because of maintenance concerns.  If revo nano was still broadly available, we'd almost certainly want it because of the form factor.  But if all we can get are clones that fit into the same footprint as existing boards, it's

davidbitton, 2016-04-08:
I ordered one and I'm going to try it out. In the end it may not be worth the development effort, but if it's not a big deal, it's sure worth trying.

mlyle, 2016-04-08:
> BTW, one point of note is the Falcon uses an MPU9250 on SPI.

Yes, that's true.  Though MPU-9250 seems to be a bit worse than 6000 when it comes to vibration sensitivity.

You may want to come by IRC as you get underway; there's a lot of people there who can answer questions.

davidbitton, 2016-04-08:
duly noted. thanks!

tracernz, 2016-04-08:
If somebody made a small form factor board with F411 (and even better, pin compatible with Revo Nano) then we'd be very interested I think. There's not much out there in small form factor ATM so it's a shame the Revo Nano never really saw the light of day.

davidbitton, 2016-04-09:
Why isn't 35mm small enough?

mlyle, 2016-04-09:
35mm is fine, but we support lots of 35/36mm flight boards-- that's the predominant form factor.  Sparky2 or Revolution etc are that size; this is lower spec than either of them.

Supporting something that's F4 (or even F3) but tiny to cram into gliders, ultramicro quads, etc, could be very nice / and is heavily differentiated from the rest of the flight boards we support.

I think you might be better off with a revolution clone-- same footprint; faster mcu with more flash; onboard radio; onboard external flash for logging; $20 less.  Revolution has a MPU-6000 on SPI, which is actually an advantage (MPU6000 seems to be more resistant to motor-PWM ultrasonics causing accelerometer bias than MPU-9xxx)

tracernz, 2016-04-09:
Basically, there needs to be something to justify the maintenance and support burden of a new target. It needs to offer something that existing targets do not.

davidbitton, 2016-04-09:
I understand now. I guess I'm accustomed to the exhaustive list of boards supported by Clean/Beta-Flight.  As for the Revolution, is [this](https://bnc.lt/m/aX1KqPEZqs) what you were referring to? I can see how it would be a preferred device.

dustin, 2016-04-09:
Does cleanflight even support stm32f4?  Their board support seems really
sparse to me, but I might just not be good at reading their docs.

I do know that many of the boards they support they support poorly.  E.g.,
you can't use the USB port on a cc3d or sparky, you can use soft serial or
a current sensor on a naze32 proper, but not at the same time.  On a
different form factor naze 32, you can't even use a current sensor at all
because you can't configure the pins.

They do support the board they designed for themselves and some of its
clones, but their HAL has been an open bug for years.

We have had a few people spin up their own flight controllers and run
dRonin on them (or just the cleanflight boards), but we don't ship support
for everything possible in the main distribution.  Some boards will give a
worse experience than others and the test matrix for official releases is
already too high.

mlyle, 2016-04-09:
@dustin summed it up pretty well.  We've been slowly doing work to clean up target support and lower the tax of supporting additional targets, but that's more so we can choose 2-3 additional targets carefully rather than open the floodgates.

@davidbitton Yes, that's the correct revolution.

davidbitton, 2016-04-09:
@mlyle ok, I'll place the order. I'll give that other F4 to my kids.