d-ronin/dRonin #904: CC3D:motor output channel assignment discontinuity

pug398, 2016-04-08:
I updated a CC3D that had a slightly unusual channel assignment in that channel 1 is dead so 2,3,4,5 are used. After updating it flies fine and mixer is set correctly except all of the motor output channel assignments say "None" in vehicle setup.

mlyle, 2016-04-09:
Just to clarify-- you didn't need to change settings-- it just displayed in the UI wrong?

I suspect this is a cousin of #315, then.

pug398, 2016-04-09:
Yes I changed nothing. It was setup and flying and I did a successful upgrade and it continued to fly. Later I noticed it had None for the Motor output channels. I have done several upgrade cycles back and forth and also manual import/export. Settings are holding now so not sure what happened.

mlyle, 2016-04-26:
I'm gonna go ahead and close this one duplicate.  I'm not sure whether it's a duplicate of #315, or #919-ish issues, or #747... (or that hopefully all of these have a common cause which seems likely)  but it almost certainly is one of these manifesting.

closed 2016-04-26 by mlyle