d-ronin/dRonin #908: GCS:SystemHealth: Audit to ensure all alarms are handled with descriptions

tracernz, 2016-04-09:
I suspect quite a few have been added without doing the necessary GCS work.

jpbarraca, 2016-04-18:
Don't know if all these are propagated to GCS. However they are set in flight but have some problem in GCS (`ground/gcs/src/plugins/systemhealth/systemhealthgadgetwidget.cpp`)

`SYSTEMALARMS_ALARM_BATTERY`
I think that Battery-Error.html is never used (but exists)

`SYSTEMALARMS_ALARM_FLIGHTTIME`
I think that FlightTime-Error.html is never used (but exists)

`SYSTEMALARMS_ALARM_GEOFENCE`
missing Geofence-Critical.html, Geofence-Warning.html and Geofence-Error.html

`SYSTEMALARMS_ALARM_GYROBIAS`
missing Gyrobias-Warning.html

`SYSTEMALARMS_ALARM_I2C`
missing I2c-Critical.html

`SYSTEMALARMS_ALARM_PATHFOLLOWER`
missing Pathfollower-Critical.html, Pathfollower-Error.html

`SYSTEMALARMS_ALARM_STABILIZATION`
missing Stabilization-Error.html

`SYSTEMALARMS_ALARM_SYSTEMCONFIGURATION`
missing SystemConfiguration-Error-NavFilter.html (not even considered in code), SystemConfiguration-Error-Multirotor.html

`SYSTEMALARMS_ALARM_TEMPBARO`
I think TempBaro-Critical.html is never used

tracernz, 2016-04-19:
Thanks @jpbarraca! Are you interested in making the missing HTML files and/or deleting the unused ones?

jpbarraca, 2016-04-19:
@tracernz I'm currently half way through #879 and can take care of this afterwards. However, this will be a busy work week and the PR can take some days to appear.

tracernz, 2016-04-19:
It's not a problem if you don't get time. I'm away from home this week but I can check it out next week if you don't get a chance.

commit 0b006bbe (2016-04-28): Cleanup Alarm descriptions. Fixes #908 (#933)

* "Cleanup Alarm descriptions. Fixes #908"

* Sync resource file with actual HTML description files
closed 2016-04-28 by mlyle