d-ronin/dRonin #910: Autoupgrader failure (cant detect bootloader/multiple bootloaders detected)

droidicus, 2016-04-10:
User on Tanto, Windows PC. Attempted to auto update, but recieved Can't Detect Bootloader and Multiple Bootloaders Detected, after choosing to go into autoupdater mode. User believes Multiple Bootloader happened before the Can't Detect Bootloader. Debug log is below.

```
10:43:48[DEBUG]"Could not open USB port"
10:43:49[DEBUG]"Could not open USB port"
10:43:49[DEBUG]"Could not open USB port"
10:43:50[DEBUG]"Could not open USB port"
10:43:50[DEBUG]"Could not open coms with the bootloader"
10:43:50[DEBUG]"GCS ready for flight board connection..."
10:43:51[WARNING]QSslSocket: cannot call unresolved function SSL_get0_next_proto_negotiated
10:43:51[WARNING]QSslSocket: cannot call unresolved function SSL_get0_next_proto_negotiated
10:43:51[WARNING]QSslSocket: cannot call unresolved function SSL_get0_next_proto_negotiated
10:43:59[DEBUG]Adding 4617 62460 1 "310024001857334231383020+BL" "LUX" "Lumenier"
10:43:59[DEBUG]usbmonitor detection cycle complete.
10:43:59[DEBUG]"More than one device was detected in bootloader state"
10:43:59[DEBUG]"Could not open USB port"
10:43:59[DEBUG]"Could not open USB port"
10:44:00[DEBUG]"Could not open USB port"
10:44:00[DEBUG]"Could not open USB port"
10:44:00[DEBUG]"Could not open coms with the bootloader"
10:44:02[DEBUG]Removing 4617 62460 1 "310024001857334231383020+BL" "LUX" "Lumenier"
10:44:02[DEBUG]usbmonitor detection cycle complete.
10:44:02[DEBUG]"Bootloader disconnection detected"
10:44:02[DEBUG]Adding 4617 62460 2 "310024001857334231383020+FW" "LUX" "Lumenier"
10:44:02[DEBUG]usbmonitor detection cycle complete.
10:44:02[DEBUG]Automatically opening device
10:44:02[DEBUG]ConnectionManager::devChanged autoconnected USB device
10:44:05[DEBUG]Connection with the autopilot established
10:44:07[DEBUG]TelemetryMonitor: connected
10:44:07[DEBUG]"Telemetry connected"
10:44:10[DEBUG]"Sending IAP Step 1"
10:44:11[DEBUG]"Sending IAP Step 2"
10:44:11[DEBUG]"Sending IAP Step 3"
10:44:12[DEBUG]"Trying to detect bootloader"
10:44:12[DEBUG]Removing 4617 62460 2 "310024001857334231383020+FW" "LUX" "Lumenier"
10:44:12[DEBUG]usbmonitor detection cycle complete.
10:44:12[DEBUG]Adding 4617 62460 1 "310024001857334231383020+BL" "LUX" "Lumenier"
10:44:12[DEBUG]usbmonitor detection cycle complete.
10:44:12[DEBUG]"More than one device was detected in bootloader state"
10:44:13[DEBUG]"Could not open USB port"
10:44:13[DEBUG]"Could not open USB port"
10:44:13[DEBUG]"Could not open USB port"
10:44:13[DEBUG]"Could not open USB port"
10:44:13[DEBUG]"Could not open coms with the bootloader"
10:44:14[DEBUG]Automatically opening device
10:44:14[DEBUG]ConnectionManager::devChanged autoconnected USB device
```

droidicus, 2016-04-10:
A different log, obtained from "save log".

[log_041016.txt](https://github.com/d-ronin/dRonin/files/211824/log_041016.txt)

tracernz, 2016-04-10:
At https://github.com/d-ronin/dRonin/blob/next/ground/gcs/src/plugins/uploader/uploadergadgetwidget.cpp#L1351 we should add (not compiled/tested):
```c++
foreach (const USBPortInfo port, devices) {
    qDebug() << QString::number(port.vendorID, 16) << ":" << QString::number(port.productID, 16) << " " << port.serialNumber << " " << port.manufacturer << " " << port.product;
}
```

droidicus, 2016-04-11:
Found the culprit, his laptop had a device called "Asus Wireless Remote Control" that was causing the GCS to see multiple bootloaders. Here are the Hardware IDs as shown by the Windows Device Manager:
```
HID\VEN_ATK&DEV_4001
HID\ATK4001
HID\*ATK4001
HID_DEVICE_UP:0001_U:000C
HID_DEVICE
```

mlyle, 2016-04-11:
Do you have the associated vendor and product id?

droidicus, 2016-04-11:
It is not a USB device, it was showing up as connected to ACPI. I think the VEN_ATK is the vendor ID in this case.

PR #915 (2016-04-14): gcs: boardmanager: don't consider vendorid 0
possibly fixes #910 ; only found through inspection.  Naze has no
vendor information--- on windows ACPI HID buttons show up with vendorid
0.
commit 8f45edf9 (2016-04-14): gcs: boardmanager: don't consider vendorid 0

possibly fixes #910 ; only found through inspection.  Naze has no
vendor information--- on windows ACPI HID buttons show up with vendorid
0.
commit 7fe1bcec (2016-04-14): gcs: boardmanager: don't consider vendorid 0

possibly fixes #910 ; only found through inspection.  Naze has no
vendor information--- on windows ACPI HID buttons show up with vendorid
0.
closed 2016-04-14 by dustin