d-ronin/dRonin #912: Flight: SUMD: Strange stick scaling on inputs

DTFUHF, 2016-04-11:
This line: https://github.com/d-ronin/dRonin/blob/080150a5b271fcfb3fe36aa1e20f92fc0d956aca/flight/PiOS/Common/pios_hsum.c#L255

has the effect of scaling SUMD's inputs around a center value of 1500, the protocol is described here on page 4: http://www.deviationtx.com/media/kunena/attachments/98/HoTT-SUMD-Spec-REV01-12062012-pdf.pdf

It's likely this scaling is used for another Graupner protocol, SUMD only requires simple /8 on these values.

tracernz, 2016-04-11:
Our convention is to not even scale channel values in the driver so this should be removed I think.

fujin, 2016-04-15:
Confirmed? I think:
[image]

tracernz, 2016-04-15:
Let's remove the scaling and pass raw values out of the rx driver, as we do for other rx drivers. ManualControl normalises them on [-1.0, 1.0] already so no point doing it twice.

mlyle, 2016-04-15:
as long as it fits easily into uint16_t and stays well over  0 and under, say, 60000, i'm cool with this plan.

tracernz, 2016-04-15:
It is a uint16_t...

tracernz, 2016-04-15:
Possible values are in the docs. I intend to replace the HSUM driver with a generic SRXL driver that supports all variants of SRXL.

mlyle, 2016-04-15:
Yes, but we have special values at the top and bottom of the uint16_t range..

tracernz, 2016-04-15:
Yep, it doesn't go to those, as per the specification document above ;-) (max is 0x41a0 == 16800). FWIW, the special values are -1, -2 and -3 (65535, 65534 and 65533).

DTFUHF, 2016-04-15:
It probably wouldn't hurt to add a check for those values upon reception,
as the protocol is capable of transmitting them.

mlyle, 2016-04-15:
If 0 is possible (haven't read the document) we should probably offset the range, too.  0 is not currently special but it seems wise to avoid it.

tracernz, 2016-04-15:
[image]

mlyle, 2016-04-15:
nice

PR #1316 (2016-09-08): hsum: eliminate scaling to 1000-2000 range
Leave it raw, so that we can have maximal resolution.

Fixes #912.

This will screw up upgrade for these users, but A) it's a small population, and B) easily fixed, and C) worth the improvement to the resolution.
commit 12075d77 (2016-09-08): hsum: eliminate scaling to 1000-2000 range

Leave it raw, so that we can have maximal resolution.

Fixes #912.
closed 2016-09-08 by tracernz