d-ronin/dRonin #914: S.Bus not working on Lux with Tanto

droidicus, 2016-04-11:
Cannot get S.Bus working on Lux FC with Tanto. If I flash Renatus it works fine, but a reflash with Tanto and no data is being received. Tried with settings translation (autoupgrade), manually, and with config wizard.

fujin, 2016-04-12:
Confirmed not working on my freshly flashed Tanto LUX :( I didn't get a chance to test before release. Doesn't pick up throttle, shows connected:false, etc.

fujin, 2016-04-12:
@mlyle maybe we wanna bump the priority on this a little? :) can we do a hotfix?

jihlein, 2016-04-12:
I see the problem for the rxport. It should work on main or flexi. I can put together the fix tomorrow evening. The port configuration is missing for the rxport. None of the serial rx's will work.

tracernz, 2016-04-12:
This is not a regression, it's been there forever: https://github.com/d-ronin/dRonin/blob/1ac0ae516f9bebbebbbeaec6ac6664bf588f5462/flight/targets/acbro/fw/pios_board.c#L308

droidicus, 2016-04-12:
@tracernz as indicated above, this works in Renatus, and does not work in Tanto, I would call that a regression in back-to-back releases.

tracernz, 2016-04-12:
Then the missing rxport config wasn't the issue? Or something else changed...

jihlein, 2016-04-12:
The missing rx port is the issue.  The problem was introduced when the PIOS_HAL argument list was reduced in PR #403.  The same issue was discovered, and fixed, in Sparky.  Unfortunately we didn't catch it in Lux.

droidicus, 2016-04-12:
@jihlein I would submit a PR to next, it can be cherrypicked if we do a hotfix at a later time.

PR #921 (2016-04-13): Flight: Lux target RX port definition
Adds uart definition to RX port.  Compiles, but needs testing by a user with a Lux board.

Fixes #914.
closed 2016-04-13 by mlyle