d-ronin/dRonin #916: CC3D upgrader failure

droidicus, 2016-04-12:
While upgrading to Tanto, friend had the following issue using the upgrader with a CC3D.
[image]

Log:
[CC3D_Updater_Log.txt](https://github.com/d-ronin/dRonin/files/214148/CC3D_Updater_Log.txt)

Then, unplugged the FC, restarted GCS, and re-plugged FC, this is the log:
[CC3D_Updater_Log_2.txt](https://github.com/d-ronin/dRonin/files/214149/CC3D_Updater_Log_2.txt)

mlyle, 2016-04-12:
may be WONTFIX-- can't do much about being unable to write to the device.

ragd011, 2016-05-08:
Just tried to update my cc3d and got the dame error.  Now it can't be detected even while booting.

mlyle, 2016-05-08:
@ragd011 start GCS, go to the firmware page, click 'rescue', then plug in the CC3D.  then when it's detected click flash and then boot.

ragd011, 2016-05-10:
OK Just tried.  I don't get the windows detect sound anymore let alone the GCS.  Rescue detected nothing.  When it was trying to upgrade it said something about a legacy bootloader install.  Right now I just get a slow blue led blink.

ragd011, 2016-05-10:
So by dropping back to the 0120 release, deleting the USB composite device and running the rescue, I was able to detect and install the 0120 firmware.  This booted successfully and the board responds normally.  It did detect once when I remove the device from device manager under the latest version but the firmware install would fail writing firmware metadata and the device would not connect after that.  Let me know if there is something else you want me to try since I think I can get back to a running device with the 0120 and 0120.3 installs.  Thanks, Rob.

mlyle, 2016-05-10:
@ragd011 -- after deleting the composite device, does it work with the current release?  If not, can you provide a log from GCS?

ragd011, 2016-05-10:
OK I eventually got it this time.  I didn't have an issue with detection this time though either.  Here's the notes and log.

installing 0409.2 over 0120.3
ran after install
connected board and got detect sound
got the message "no valid firmware on device"
Clicked yes
this launched the autoupgrade
automatic upgrade failed - unable to flash upgrader image to board!
still got usb sound when disconnect and reconnect
this time clicking rescue it was able to do something but it wasn't saying what
then it showed that it was in the bootloader so I clicked flash and it did the erase and firmware/metadata flash
[cc3d_upgrade.txt](https://github.com/d-ronin/dRonin/files/258057/cc3d_upgrade.txt)

mlyle, 2016-05-10:
Does it work after flashing the board manually?

ragd011, 2016-05-11:
Ya sorry, forgot to mention, in the end it's running the current release and gets detected by the GCS fine.  I did notice the time it takes to go from the bootloader detect to the running firmware is much faster.  I'm guessing that was the new bootloader that was flashed during the update?

mlyle, 2016-05-11:
No, no new bootloader is flashed for CC3D; but GCS will tell the bootloader to exit early.

Thanks for the report and I'm glad you made it to the new release.

ragd011, 2016-05-11:
Just got back from an autotone flight.  It went pretty well for the most part.  I landed, disarmed, pulled the power and went inside.  Then I connected to the my Win10 machine and connected to the FC.  I saw the calculated values and saved them to the stabilization page.  I unpluged and re-connected to make sure everything stayed.  Then I changed the mode switch to acro from autotune for pos 2, disabled autotune module and saved/applied everything.  This way I could go back in and enable BT Telem.  So I disconnected and reconnected to check everything was good before enabling BT and I can't get to the board again.  I'll have to mess with it tomorrow. So Close.

ragd011, 2016-05-11:
OK tried my win 10 laptop and it works fine there.  It detects the FC and has the changes stored.  I connected and disconnected a few times.  So maybe soemthing with my other machine.  I'll try to use both before I post issues.  Thanks, Rob.

mlyle, 2016-06-12:
I think this is either hardware or covered by another bug; so closing.

closed 2016-06-12 by mlyle