d-ronin/dRonin #919: Report: some imported values don't get saved

mlyle, 2016-04-12:
Unconfirmed, but reported by @pug398 who is distressingly accurate about these types of things.

Particularly Altitude Settings or Sparky1 HW settings.

It's possible we have some kind of issue with the save-object state machine on GCS or flight side, or it's a general problem with telemetry.  Or something else entirely.

pug398, 2016-04-12:
I will watch the update process closer and name each fc cloud settings file with a unique identifier. Then I will still have the original exported data should I find an issue after doing multiple auto updates. The previous reported problem of PathPlanner settings not importing to sparky2 flash might be related to this.

pug398, 2016-04-15:
Sparky2 simply does not save AltitudeHold settings in flash for me. They import in but do not save to flash. They are present in the cloud file and exported uav file, import in (both auto/manual) but as soon as board power is cycled it is lost. You can import either way, go to the browser, save and they stick. It is not being written to flash on import.

pug398, 2016-04-16:
20160211-efc3 and previous (working), 20160223-49ed and later (not working). Most likely introduced when import changes were made somewhere in between.

mlyle, 2016-04-16:
@pug398 cool-- that was one of two things I suspected.

Kids are off school this week so I've not been super productive-- but I'll be able to audit that set of importer changes soon to see why it could be skipping saves.

pug398, 2016-04-24:
With 685711b0 20160418 HW, AH and PathPlanner settings all saved permanently on sparky2.
All settings saved permanently on sparky1. The ones I had issue with appear to be saving now.

mlyle, 2016-04-24:
Ick.. we didn't fix anything :P

pug398, 2016-04-24:
Perhaps a change in firmware timing? Updating firmware was the only change on my end.

jihlein, 2016-04-25:
I'm seeing some odd behavior on latest next with AQ32 target.  I can't save frame type or output channels.  I did an erase settings and tried again, still unable to save.

Also have noticed the same on Sparky.

mlyle, 2016-04-25:
@jihlein -- try a clean build.  Odds are there's problems with build dependencies resulting in mismatched objects between firmware and GCS to get problems like that.

jihlein, 2016-04-25:
I did.  Just noticed that if I hit save twice before rebooting, it saves.  Let me experiment some more.

pug398, 2016-04-25:
I have seen the need to save twice before as well. c98a wouldn't save some objects but 6857 has saved all objects so far. As soon as Jenkins is going again I will try the next build to see if it continues to save or whether it was just blind luck.

pug398, 2016-04-26:
I just upgraded sparky1, sparky2, revo and quanton with a7dc and all settings saved properly. Don't know but it sure seems a lot better now :)

mlyle, 2016-07-22:
Closing because no recent repro.  Yuck.  (Though there's been some related fix / robustness improvement).

closed 2016-07-22 by mlyle