d-ronin/dRonin #924: Quanton: support USART6 on rcvrport

mlyle, 2016-04-13:
More serial is always good, right?

Unfortunately the CLK line for it is used for I2C; else we could have used it to get a pseudo-SPI out.

jihlein, 2016-04-13:
PWM IN2 would function as USART6 TX, PWM IN3 would function as USART6 RX.

I would suggest adding 3 more selections to the RCVR port selection:

PPM+Serial
PPM+Serial+ADC
Serial

If desired, serial could also be added with the PWM output function, but the permutations of pin use start to get a little silly in my opinion.

USART6 port selection would be the typical cast of characters.

mlyle, 2016-04-13:
I agree re: permutations.

Any strong reason to not have PPM+Serial+ADC only?

jihlein, 2016-04-13:
Not really.  I already have the three selections mentioned above ready for testing, but can easily pare them down if that's the desired path.

mlyle, 2016-04-13:
Awesome :)  Whatever you think is best.  Thanks for just fixing everything always ;)

tracernz, 2016-05-01:
Done.

closed 2016-05-01 by tracernz