d-ronin/dRonin #928: STM32 Libraries not GPL compatible

mlyle, 2016-04-17:
Need to add appropriate exception to our license to allow linking with STM32 libraries.  Need to make best-effort to clear with existing contributors.  (But should be OK).  Need to be careful going forward-- it'll be difficult to bring in any third-party GPL code.

mlyle, 2016-04-17:
The issue is this clause:

```
4.	This software, including modifications and/or derivative works of this software, must execute solely and exclusively on microcontroller or microprocessor devices manufactured by or for STMicroelectronics.
```

which is an incompatible restriction on use.

The GPLv3 says:
```
A “Standard Interface” means an interface that either is an official standard defined by a recognized standards body, or, in the case of interfaces specified for a particular programming language, one that is widely used among developers working in that language.

The “System Libraries” of an executable work include anything, other than the work as a whole, that (a) is included in the normal form of packaging a Major Component, but which is not part of that Major Component, and (b) serves only to enable use of the work with that Major Component, or to implement a Standard Interface for which an implementation is available to the public in source code form. A “Major Component”, in this context, means a major essential component (kernel, window system, and so on) of the specific operating system (if any) on which the executable work runs, or a compiler used to produce the work, or an object code interpreter used to run it.
```

The ST libraries probably do not qualify under the system library exception as written.

But I think this is the best path-- to clarify that we consider the STM32 libraries and CMSIS components included within the source as "System Libraries" under the terms of the GPL, in our licensing document and contributing.md file.

mlyle, 2016-04-17:
http://debian.2.n7.nabble.com/MCD-ST-Liberty-SW-License-Agreement-td3526097.html

tracernz, 2016-04-17:
There is a 3rd party GPL ST lib out there.

mlyle, 2016-04-17:
If it's this thing -- https://github.com/MaJerle/stm32fxxx_hal_libraries --- it's ostensibly under the GPL but contains the ST code :/

PR #1015 (2016-05-29): LICENSE: update to mention stdperiph.
Fixes #928.
commit 133fec2f (2016-05-29): LICENSE: update to mention stdperiph.

Fixes #928.
closed 2016-05-29 by mluessi