d-ronin/dRonin #930: Request: detect crash, stop "motors spin at neutral"

reubenfirmin, 2016-04-17:
* Seen in version (release version or githash): 4/09
* On flight controller (if applicable): CC3d

Issue details: "Motors spin at neutral" is a cool feature, but please somehow detect crashes and disable spinning when that happens. Having motors spinning in long grass is not good. Yes, disarming is the answer, but it may be all that the user can do to kill throttle after the crash.

mlyle, 2016-04-18:
Open for comments.

My thoughts are:

* I can't reliably detect a crash.  Big accelerometer transients etc are not enough / happen in normal flight / landing / etc.
* Hitting gates is usually a huge transient but often recoverable and something our users often want.
* Spuriously disarming is really bad.

It's possible to set ArmedTimeout to smaller values, and there is the still undocumented / unexposed in UI "ManualControlCommand.InvalidRXArmedTimeout" which can be used to disarm more quickly when we're not receiving commands from the receiver.

mlyle, 2016-04-18:
(P.S. If your intention is to just disable "spin at armed" and not disarm-- I am strongly opposed.  That means that the quad will still be armed and just ready to chew up someone's face if they bump the throttle-- but concealed / look exactly like it does when disarmed).

closed 2016-05-18 by mlyle