d-ronin/dRonin #935: Max GPS sattelites - make it configurable

bazyl78, 2016-04-25:
Issue details:

Please make configurable max number of sattelites like it is in arducopter. It takes ages for a small gps on micro quadcopter (Ublox 7 Series Mini GPS) to wait for reaching 7 sats count. Would be great to be able to decrease it to 5-6. Or at least don't block arming and just not allow to switch into position hold mode.

Code to be changed:
// A more stringent requirement for GPS to initialize the filter
	bool gps_init_usable = gps_updated && (gpsData.Satellites >= 7) && (gpsData.PDOP <= 3.5f) && (homeLocation.Set == HOMELOCATION_SET_TRUE);

Just make it not hardcoded and set a variable visible in UAV settings.

DRUMSZ, 2016-05-26:
how or did you do this ??
same issue but can't reload libre.......
Thanks

mlyle, 2016-05-27:
Dup / strongly related #155 ... I agree it's unfortunate.

closed 2016-05-27 by mlyle