d-ronin/dRonin #969: GCS: Telemetry crash

tracernz, 2016-05-05:
During upgrader, thread #1 segfaulted by de-referencing `sessionRetrieveTimeout  0x0 QTimer *` in `TelemetryMonitor::retrieveNextObject`:
```
		
	Locals		
		obj	0x3	UAVObject *
		this	<not accessible>	TelemetryMonitor
			[QObject]	@0x38c8ce0	QObject
			CONNECTION_TIMEOUT_MS	8000	int
			STATS_CONNECT_PERIOD_MS	350	int
			STATS_UPDATE_PERIOD_MS	1600	int
			connectionStatus	TelemetryMonitor::CON_DISCONNECTED (0)	TelemetryMonitor::connectionStatusEnum
			connectionTimer	0x0	QTime *
			delayedUpdate	<not accessible>	QList<UAVDataObject *>
			flightStatsObj	<not accessible>	FlightTelemetryStats
				[UAVDataObject]	@0x7ffff65130e0	UAVDataObject
				CATEGORY	""	QString
				DESCRIPTION	"Maintains the telemetry statistics from the OpenPilot flight computer."	QString
				FIELD_DESCRIPTIONS	<6 items>	QHash<QString, QString>
				ISSETTINGS	false	bool
				ISSINGLEINST	true	bool
				NAME	"FlightTelemetryStats"	QString
				NUMBYTES	21	quint32
				OBJID	796797186	quint32
				data	@0x7ffff651313c	FlightTelemetryStats::DataFields
				staticMetaObject	@0x7fffc0974880	QMetaObject
				[11]		
			gcsStatsObj	<not accessible>	GCSTelemetryStats
				[UAVDataObject]	@0xce7c00	UAVDataObject
				CATEGORY	""	QString
				DESCRIPTION	"The telemetry statistics from the ground computer"	QString
				FIELD_DESCRIPTIONS	<6 items>	QHash<QString, QString>
				ISSETTINGS	false	bool
				ISSINGLEINST	true	bool
				NAME	"GCSTelemetryStats"	QString
				NUMBYTES	21	quint32
				OBJID	2881955652	quint32
				data	@0xce7c5c	GCSTelemetryStats::DataFields
				staticMetaObject	@0x7fffc0974a40	QMetaObject
				[11]		
			isManaged	false	bool
			numberOfObjects	0	quint8
			objMngr	<not accessible>	UAVObjectManager
				[QObject]	@0x77cba0	QObject
				MAX_INSTANCES	1000	quint32
				objects	<not accessible>	QHash<unsigned int, QMap<unsigned int, UAVObject *>>
				objectsByName	<not accessible>	QHash<QString, QMap<unsigned int, UAVObject *>>
				staticMetaObject	@0x7fffc0972cc0	QMetaObject
				[5]		
			objectRetrieveTimeout	0x0	QTimer *
			queue	@0x38c8d10	QQueue<UAVObject *>
				[QList<UAVObject *>]	<not accessible>	QList<UAVObject *>
			retries	0	int
			sessionID	0	quint16
			sessionInitialRetrieveTimeout	0x0	QTimer *
			sessionObj	0x0	SessionManaging *
			sessionObjRetries	0	int
			sessionRetrieveTimeout	0x0	QTimer *
			sessions	<not accessible>	QHash<unsigned short, QList<TelemetryMonitor::objStruc>>
			settings	0x21	Core::Internal::GeneralSettings *
			staticMetaObject	@0x7fffbb3a7980	QMetaObject
			statsTimer	0x0	QTimer *
			tel	@0x11d4680	Telemetry
				[QObject]	@0x11d4680	QObject
				MAX_QUEUE_SIZE	20	int
				MAX_RETRIES	2	int
				MAX_UPDATE_PERIOD_MS	1000	int
				MIN_UPDATE_PERIOD_MS	1	int
				REQ_TIMEOUT_MS	250	int
				gcsStatsObj	0x4026000000000000	GCSTelemetryStats *
				objList	<not accessible>	QVector<Telemetry::ObjectTimeInfo>
[... 586 lines cut ...]
        app = <incomplete type>
        dirName = {static null = {<No data fields>}, d = 0x89a3f0}
        locale = {static null = {<No data fields>}, d = 0x96fec0}
        parser = {d = 0x899fd0}
        helpOption = {d = {d = 0x8994b0}}
        versionOption = {d = {d = 0x899960}}
        showVersionOption = {d = {d = 0x8996e0}}
        overrideOption = {d = {d = 0x89ad60}}
        cleanConfig = {d = {d = 0x89a7c0}}
        noLoadOption = {d = {d = 0x89a9c0}}
        doTestsOption = {d = {d = 0x89d730}}
        pluginOption = {d = {d = 0x89d930}}
        settingsFilename = {static null = {<No data fields>}, d = 0x89dcf0}
        positionalArguments = {<QList<QString>> = {<QListSpecialMethods<QString>> = {<No data fields>}, {p = {static shared_null = {ref = {atomic = {_q_value = {<std::__atomic_base<int>> = {static _S_alignment = 4, _M_i = -1}, <No data fields>}}}, alloc = 0, begin = 0, end = 0, array = {0x0}}, d = 0x7ffff6513130 <QListData::shared_null>}, d = 0x7ffff6513130 <QListData::shared_null>}}, <No data fields>}
        translator = <incomplete type>
        qtTranslator = <incomplete type>
        pixmap = <incomplete type>
        splash = {<QSplashScreen> = {<No data fields>}, static staticMetaObject = {d = {superdata = 0x7ffff7930960 <QSplashScreen::staticMetaObject>, stringdata = 0x438c60 <qt_meta_stringdata_CustomSplash>, data = 0x438d40 <qt_meta_data_CustomSplash>, static_metacall = 0x412bc4 <CustomSplash::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)>, relatedMetaObjects = 0x0, extradata = 0x0}}, m_progress = 100, m_pixmap = <incomplete type>, m_progress_bar_color = {cspec = QColor::Rgb, ct = {argb = {alpha = 65535, red = 55769, green = 30326, blue = 0, pad = 0}, ahsv = {alpha = 65535, hue = 55769, saturation = 30326, value = 0, pad = 0}, acmyk = {alpha = 65535, cyan = 55769, magenta = 30326, yellow = 0, black = 0}, ahsl = {alpha = 65535, hue = 55769, saturation = 30326, lightness = 0, pad = 0}, array = {65535, 55769, 30326, 0, 0}}}, time = {mds = 70725670}, message_number = 215, settings = <incomplete type>}
        gcsTranslationsPath = @0x7fffffffdea0: {static null = {<No data fields>}, d = 0x7fc290}
        pluginManager = {<QObject> = {<No data fields>}, static staticMetaObject = {d = {superdata = 0x7ffff68c31c0 <QObject::staticMetaObject>, stringdata = 0x7ffff797be00 <qt_meta_stringdata_ExtensionSystem__PluginManager>, data = 0x7ffff797bfa0 <qt_meta_data_ExtensionSystem__PluginManager>, static_metacall = 0x7ffff797880c <ExtensionSystem::PluginManager::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)>, relatedMetaObjects = 0x0, extradata = 0x0}}, d = 0x812450, static m_instance = 0x7fffffffe1a0, m_lock = {d = 0x9664f0}, m_allPluginsLoaded = true}
        pluginPaths = {<QList<QString>> = {<QListSpecialMethods<QString>> = {<No data fields>}, {p = {static shared_null = {ref = {atomic = {_q_value = {<std::__atomic_base<int>> = {static _S_alignment = 4, _M_i = -1}, <No data fields>}}}, alloc = 0, begin = 0, end = 0, array = {0x0}}, d = 0x801800}, d = 0x801800}}, <No data fields>}
        parsingErrors = {<QList<QString>> = {<QListSpecialMethods<QString>> = {<No data fields>}, {p = {static shared_null = {ref = {atomic = {_q_value = {<std::__atomic_base<int>> = {static _S_alignment = 4, _M_i = -1}, <No data fields>}}}, alloc = 0, begin = 0, end = 0, array = {0x0}}, d = 0x7ffff6513130 <QListData::shared_null>}, d = 0x7ffff6513130 <QListData::shared_null>}}, <No data fields>}
        plugins = {<QListSpecialMethods<ExtensionSystem::PluginSpec*>> = {<No data fields>}, {p = {static shared_null = {ref = {atomic = {_q_value = {<std::__atomic_base<int>> = {static _S_alignment = 4, _M_i = -1}, <No data fields>}}}, alloc = 0, begin = 0, end = 0, array = {0x0}}, d = 0x7fd3d0}, d = 0x7fd3d0}}
        coreplugin = 0x7bf3e0
```

mlyle, 2016-05-16:
It looks like several of the deletes in uavtalk should be deletelaters.

Does this repro reliably for you?  I have never seen it.

mlyle, 2016-05-16:
(I can make the change if you are able to see if it cures stuff).

mluessi, 2016-05-16:
I also have noticed the GCS segfaulting frequently. Most recently when trying to enter the bootloader. Probably related.

mlyle, 2016-05-16:
For me using Windows and OSX GCS has been much more reliable than ever before, but there are definitely some new defects (e.g. #998).  I used to get many crashes per day but now it's on the frequency of one every few days.

mluessi, 2016-05-16:
Something is definitely broken on Linux. I also just had the uploader fail ("Firmware metadata upload failed"). This is on brain / next, so unrelated to RE1.

tracernz, 2016-05-16:
> I also have noticed the GCS segfaulting frequently. Most recently when trying to enter the bootloader. Probably related.

Yeah, that's when I see it most times.

tracernz, 2016-05-17:
Added BLOCKER, this prevents the upgrader about 90% of the time.

PR #870 (2016-05-24): gcs: hidapi: use rawhid on Linux, instead of libusb
Better performance, simpler, thunks directly to udev, constructs
meaningful paths.

Eliminates dependency on libusb.

Move to use paths in rawhid for connection (saving re-enumeration and associated risks) as part of this branch.

Use queued connections for check-for-upgrade behavior-- fixes #969 fixes #1010

Errata/relnotes: REQUIRES NEW UDEV RULES
closed 2016-05-24 by tracernz