d-ronin/dRonin #970: TxPID not usable - extremely large values

fabianhu, 2016-05-05:
* Seen in version (release version or githash): 2016-04-09.2
* On flight controller (if applicable): Sparky
OS: Ubuntu 15.10

Issue details:
Set up TxPID, with SUMD as input, selected aux 0..2 as inputs
entered e.g. 10 and 50 as the range values, which are converted to 10,00000 and 50,00000 (where ',' (Comma) is the decimal separator in my locale
Everything seems fine to this point.

In the Stability pane, the PID values are updated with values 10000... (can't see all zeroes) to 50000....
So there seems some issue with the decimal separator  processing.

mlyle, 2016-05-05:
@fabianhu the "real" PID values-- as shown on the advanced and expert panes are the actual values multiplied by the angular rate or the angle to come up with control outputs.  These are very small values--  the default P of 0.002 reacts to an angular rate of 10 deg/sec with a 2% control movement.

However, values like 0.00003 for derivative aren't very user friendly to someone sliding a big slider, so the basic tab has values that are significantly higher-- 10000x higher.

mlyle, 2016-05-05:
P.S. the scaling factor for basic kD is: ```<string>scale:0.000001</string>```

closed 2016-07-22 by mlyle