d-ronin/dRonin #971: Feature: Possibility in GCS to see which settings are different than the defaults

yamahabest, 2016-05-05:
* Seen in version (release version or githash): -
* On flight controller (if applicable): -

Issue details:

I just understood from fuijn at IRC that the default value for Motor curve fit changed from 1.0 to 0.9 in the latest release. I upgraded however from an older release to the new release, and still had the old value of 1.0.

I understand that it is not possible to update this setting (and other settings) automatically during upgrade, as it could be that the user needs it to be at that value.
However, users currently don't know that the defaults have been changed.

It would be nice to have an overview of the settings, with a comparison against the default values (in that specific release of the GCS), with the ability to set the different value to that default value.

Ps. There is of course the "Erase all settings from board", however, that also erases the Input/Output stuff, which is less convenient.

tracernz, 2016-05-05:
We could do something through the UAVO->Widget relation mechanism for this. It's just a question of what we want to actually do. Should probably do it in the UAVO browser too.

tracernz, 2016-10-09:
So, this is done in UAVO browser already (tiny part of #1279).

yamahabest, 2016-10-18:
Would it be possible to detect that a value was never changed from the default value?
If so, you could highlight the field (with a color, or an informational "i" icon), without highlighting the "false-postives" (the fields that have been changed by the user).

tracernz, 2016-10-18:
Well, hopefully not that many things need to be changed from default so I think that would be way too noisy. :wink:

yamahabest, 2016-10-19:
Could be, yes. Just trying to find a workable solution.