d-ronin/dRonin #972: Feature: TX/RX/Error/Warnings/Info widget: Show errors and warnings in tooltip and/or enable clicking on it, which will take the user to the Flight data or Firmware tab

yamahabest, 2016-05-05:
Currently, when there are errors or warnings, the users gets the blinking icon(s).
The user can hover over them, and get some cryptic tooltip messages: "_blabla_ is in Error state", which doesn't help the user.

It would be nice if the tooltip would show the actual errors/warnings in that tooltip and/or the user is able to click that error or warning icon, and will be taken to the System Overview widget on the Flight data or Firmware tab.

tracernz, 2016-05-05:
Is this in the bottom bar?

yamahabest, 2016-05-05:
Yep, was trying to indicate that one.

tracernz, 2016-05-05:
So clicking it should find a tab with the system health widget and switch to it? Should be doable. Hopefully could pop up the error descriptions as well.

yamahabest, 2016-05-05:
Yes, something like that. Would be helpful for new users.

tracernz, 2016-05-05:
Yep, thanks for the report! :-)

mlyle, 2016-09-08:
Ideally, I think hovering should just show the tooltip text that the system health widget would show us, but dunno how tricky that is.