d-ronin/dRonin #974: Feature: Check from GCS for new release, and offer to update

yamahabest, 2016-05-05:
Could help getting users to the latest release.

tracernz, 2016-05-05:
@PTDreamer was working on this but he's been MIA for a while..

yamahabest, 2016-05-08:
That's too bad. Functionality would be nice to have.

mlyle, 2016-05-09:
Yup.

Whether there's auto-upgrade (which I'm a bit meh about) or notification... it would be nice to have.

It pays no dividends until the release after it's shipped, though. ;)

tracernz, 2016-05-09:
Notification should be easy enough. Just a question of whether we do it through autotown or directly with github API.

yamahabest, 2016-05-09:
I could imagine an Alpha, Beta, Release setting.
Where Alpha is "nightly" build, Beta pre-release and Release ...

Would really like auto-upgrade, but a notification (maybe with the release notes, and a link to the download location) is already a good step.

dustin, 2016-05-15:
There was discussion about this yesterday which seemed to conclude that using something like an [appcast](https://github.com/vslavik/winsparkle/wiki/Appcast-Feeds) would be appropriate.

PR #1051 (2016-06-23): Add release announcements to welcome plugin
Fixes #974

How it works:

We download a file from `http://dronin.org/relAnnounce/<<githash>>.json`

If it exists, and looks like:

```
{ 
"thresh" : 700, 
"text" : "<h1><font color=\"red\">Oh Shit</font></h1><p>Better upgrade now.  Get it <a href=\"http://dronin.org\">here</a>.  Fast.</p><p>A <font color=\"red\">CRITICAL</font> bug has been found that will make your quadcopter catch fire.</p>"
}
```

We compare thresh to a hash of our usage info.  Basically, thresh is the per-1000 number of users that should be notified (e.g. if it's 500, half of us [... cut]
closed 2016-06-23 by tracernz