d-ronin/dRonin #985: Corner twitch/pull on current-generation invensense parts.

mlyle, 2016-05-09:
Possible issues seen with ESC ultrasonic noise getting into gyros and "twitching" FC.

Seems tied to derivative term.  Seen so far on Naze32R6 (MPU-6500) and Brain (MPU-9250).  Eliminating D term on brain cured twitch; @ufoDziner is going to try now with derivative cutoff set to 20 vs. old value of 37.

mlyle, 2016-05-09:
Happened with d=0.  Closing invalid.

closed 2016-05-09 by mlyle

tracernz, 2016-05-09:
Alternatively you can adjust derivative gamma and see if the feel is okay.

mlyle, 2016-05-09:
At first we thought D=0 was curing it; and were playing with derivative cutoff (not gamma)...

But eventually @ufoDziner got a repro with D=0, so it doesn't have much to do with the derivative controller.