d-ronin/dRonin #988: FlySky i-bus protocol support

fujin, 2016-05-11:
* Seen in version (release version or githash): 
All

* On flight controller (if applicable):
All

Issue details:
FlySky i-bus protocol is not available under serial port options. No FlySky i-bus protocol appears to be implemented. User with FlySky i-bus protocol radio only cannot make use of dRonin, requires switching to PPM or changing radio system.

fujin, 2016-05-11:
I have heard that it may be possible to use `ni-sbus` for this, but haven't investigated at all. I am looking at getting some i-bus gear from local supplier, it comes up pretty often in our beginner groups.

dustin, 2016-05-11:
From what I can tell, it's very different from s.bus.  The CF reverse engineering project is the closest thing I can find to a protocol spec: https://github.com/cleanflight/cleanflight/issues/1125 (I guess spec availability is a similarity to s.bus).

tracernz, 2016-05-12:
https://github.com/multiwii/baseflight/pull/241

bazyl78, 2016-05-13:
This would be really helpfull. There is arduino code ibus2ppm if it helps.

tracernz, 2016-05-13:
@bazyl78 Check out #991

bazyl78, 2016-05-16:
Great man :)

PR #991 (2016-06-21): IBus Rx Driver
First cut, totally untested, based on original baseflight implementation.

Needs:
* [x] Doxygen docs
* [x] Test on FlySky hardware
* [x] Added to more targets

Fixes #988
closed 2016-06-21 by mlyle