d-ronin/dRonin #995: Reverse motor directions Quad X motor layout

emil-nasso, 2016-05-15:
Is it possible to set up a Quad X motor layout with dRonin?

There are two benefits to this, as I see it:

* If you fly into grass/vegetation, it will be thrown away from the camera instead on spraying the lens with chopped up grass. :) I can't be the only one that sucks at racing and spend way too much time in tall grass and has a frame/camera that is completely covered in grass from time to time? ;)
* If you hit something with a prop (for example a race gate) the prop rotates "with the collision" instead of rotating into the item you collide with. This could perhaps help a bit with recovery but primarily, it should be a bit less friendly towards the motors.

Is there a way to do this in dRonin?

Being able to choose "Quad X - reversed direction" frame type would be a really cool feature, in my opinion!

tracernz, 2016-05-15:
Already been possible for many years. ;-)
[image]

emil-nasso, 2016-05-15:
Holy shit. I'm embarrassed, and slightly blind, apparently.... :)

Move along, nothing to see here.

closed 2016-05-15 by emil-nasso

tracernz, 2016-05-15:
No problem! :-)