d-ronin/dRonin #999: ADC RSSI only works for ADC0 and ADC1

mluessi, 2016-05-16:
For some reason, analog RSSI only works for the first two ADC inputs. Tested on Brain and RE1. The same pin / ADC works in the battery module. 

Not sure what the reason is.

mluessi, 2016-05-16:
Also confirmed on Sparky2

tracernz, 2016-05-18:
Same code, with the exception of the scaling, is used by battery (GetChannelVolt) and RSSI(GetChannelRaw).
https://github.com/d-ronin/dRonin/blob/next/flight/PiOS/Common/pios_adc.c#L184-L229

mluessi, 2016-05-18:
@tracernz found the problem: When selecting "ADC0", the GCS sets the `RssiChannelNumber` to 1 instead of 0, so the ADC inputs are always 1 off. Note that this is also true for other RSSI inputs. I suggest we remove the "Disabled" option from the combo box, as it is redundant with RSSI type `None`. WDYT?

tracernz, 2016-05-18:
Good catch. The "Disabled" option applies to all instances of InputChannelForm, and special-casing RSSI would make it inconsistent with the other channel types. We should just subtract 1 I think?

mluessi, 2016-05-18:
@tracernz yes, your solution is better. For other RSSI types it was actually correct as for the receiver types channel 0 means disabled.

PR #1009 (2016-05-19): FIX: RSSI ADC channel
Fixes #999 

Tested on Brain.
closed 2016-05-19 by mlyle