d-ronin/dRonin #1003: pios_pwm/pios_ppm ignore timer resolution

tracernz, 2016-05-17:
Since 12 MHz was originally introduced. This is a problem on targets which share a timer between inputs and outputs such as the CC3D.

mluessi, 2016-05-20:
@tracernz how about adding a "max clock" option to the timer config? It could be used to limit the timer clock on shared timers on CC3D.

tracernz, 2016-05-21:
That would limit serial RX users to 1 MHz outputs unnecessarily though.

mluessi, 2016-05-25:
@tracernz we could use a different timer init struct depending on the Rx selected.

tracernz, 2016-05-25:
Might as well just scale to us properly based on the timer clock IMO. Would be easier and one less target-specific thing needed.

mluessi, 2016-05-26:
Yeah, but isn't the problem of having a very fast running shared timer that it will cause a ton of overflow interrupts when it is also used for input?

tracernz, 2016-05-26:
It's not compatible with SyncPWM at all, so we need only consider traditional PWM cases. 1500 us servos/ESCs will be fine, and 760us servos should be fine. Brushed output is the problematic case, and it's pretty much incompatible. Either you have very poor output resolution, or lots of overflow interrupts for the input.

mlyle, 2016-06-09:
IMO this was adequately solved by #958.

closed 2016-06-09 by mlyle