d-ronin/dRonin #1005: CC3D not seeing particular input channel

reubenfirmin, 2016-05-18:
* Seen in version (release version or githash): 2016-04-09
* On flight controller (if applicable): CC3D

Issue details:
Symptom - a particular channel isn't seen by CC3D/Dronin in the radio setup wizard.
 
I've tried the following:
1) Two different radios and two different receivers (Taranis & Flysky); both exhibit the same issue (the channel that isn't seen differs if I juggle around the cables going into the receiver).
2) Three different CC3D cables. All exhibit the same behavior.

I don't recall the history of this CC3D. It most likely came from another build (in which case it worked perfectly well in that build), but there's the offchance it was new on the shelf. So, it may have had a prior version of Dronin on it, or have been a full upgrade from Openpilot.

In the manual settings, Dronin only lets me set up channels 1-5; choosing channel 6 gets blanked out after a short period of time. In the wizard, I step through until one or other of the controls doesn't work.

It's entirely possible this is a hardware issue; is there any diagnostic dronin can do to prove that? (If not, this is a feature request.)

mlyle, 2016-05-18:
First: I don't think issues are an appropriate place to seek support.  It clutters the issue database, and we have 3 other mechanisms stood up for this-- the google group, the facebook page, and the IRC channel.

If a channel doesn't work, I'm guessing that you're trying to use oneshot on an output bank or 12MHz timer resolution.  This is documented to not work because of hardware limitations of CC3D.:

https://github.com/d-ronin/dRonin/wiki/User-Guide:-OneShot125#special-note-for-cc3d

> In the manual settings, Dronin only lets me set up channels 1-5; choosing channel 6 gets blanked out after a short period of time. In the wizard, I step through until one or other of thcontrols doesn't work.

This sounds like you were in calibration mode (which doesn't have the best UI)-- and prevents updates to configuration elements.  (It's for detecting the limits and centers of the channels).

closed 2016-05-18 by mlyle

reubenfirmin, 2016-05-18:
No, I'm not trying to use Oneshot, and haven't done anything special regarding 12MHz.

And I've tried both manual and the wizard. Neither can see more than 4 channels; that's what I was trying to describe.

I filed a bug because I think it's highly unlikely to be a hardware issue. As I described, I've eliminated the radio and the wiring as being potential contributors to the problem. That leaves the CC3D, and its firmware. I can't imagine why one channel on the hardware wouldn't work -- but if that is the case, that leads to my feature request ticket: diagnostics. Essentially dronin doesn't give me any way to debug further.

mlyle, 2016-05-18:
I recommend you utilize normal support channels and not leap to open issues.

This lets other people of the community answer your questions besides the core developers who are trying to solve things from here, and the actual developers get stuff done.

It also doesn't clutter up the database with things that are not defects.

Things that come up with multiple users experiencing them get opened as usability defects here.  Things that are triaged as genuine defects are opened here.

> I filed a bug because I think it's highly unlikely to be a hardware issue.

Well... except all the PWM channels are working for everyone else... we have quite a few people on CC3D running arming switch + flightmode configurations using all the PWM in...  Like the first CC3D with a shared tune on autotown, for instance: http://dronin-autotown.appspot.com/api/tune?tune=ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICAwOn8hQkM

>  I can't imagine why one channel on the hardware wouldn't work -- but if that is the case, that leads to my feature request ticket: diagnostics. Essentially dronin doesn't give me any way to debug further.

There's a whole uavobject browser that exposes the entire internal state of the system on the right side of the screen in config mode.  Every bit of communication between different modules happens through these objects and is fully exposed.  There is some documentation for it, and there's many moderately advanced users who are using it in practice (and giving guidance to people who need help troubleshooting problems on IRC / the google group).