d-ronin/dRonin #1006: CC3D bricked during upgrade from OpenPilot

reubenfirmin, 2016-05-18:
* Seen in version (release version or githash): 2016-04-09
* On flight controller (if applicable): CC3D

Issue details:
I have an Eturbine (basically a metal Eachine 250 with some higher quality build elements.) It's been on OpenPilot since I got it, and I finally decided to upgrade it to Dronin and autotune it.

I plugged in the USB cable, and Dronin detected foreign firmware, and brought up the auto-upgrade dialog. It ran through to the final step, which IIRC has to do with settings. It then popped up an error dialog saying something like "Failed to copy settings; [perhaps something like no settings detected", and gave me the options to Abort or Cancel (again moderate IIRC). I pressed Abort, and which point it popped up a dialog saying that the upgrade failed. After this it was bricked; the power LED comes on when I plug it into USB, but the sentience LED never flickers. 

I'm happy to mail this to somebody if you'd like to do some diagnostics on it to find out what went wrong. And/or I can poke around if you give me instructions.

mlyle, 2016-05-18:
Try the rescue process-- click rescue then plug in board.  You can flash firmware manually then.

It's hard to know what happened without knowing exactly what you clicked on (I don't think the particular combinations of clicks in this issue are possible) and without a debug log.  Asking it to try and import settings from openpilot definitely doesn't work (but also doesn't leave the board in a stuck state).

If that doesn't work, there's either something wrong with the computer being used with GCS (a few people seem to have USB communications issues) or hardware on the individual CC3D.  The bootloader runs before the firmware; and the bootloader is not capable of overwriting itself.

closed 2016-05-18 by mlyle

reubenfirmin, 2016-05-18:
I will try rescue and get back to you in a week or so on this. It's not the computer; I've done literally dozens of CC3D installs using it. 

Where is the debug log? That would be useful to know if it happens again.

Shouldn't the wizard _not offer_ to import settings if it knows it can't?

tracernz, 2016-05-18:
> Shouldn't the wizard not offer to import settings if it knows it can't?

Yep, been discussed for situations where no settings are present, but not a top priority.

mlyle, 2016-05-18:
It pops up a message that it can't identify the firmware on the flight board, and asks you whether you want to clear the board, or try to import the settings like it was our last release.

There are people running custom firmwares that our cloud upgrader doesn't know about but can possibly upgrade successfully from.  There are also people running openpilot.  We can't tell the difference.