d-ronin/dRonin #1007: Lux: Disable PPM if Serial RX configured

tracernz, 2016-05-18:
#1000 restored the behaviour of always initing PPM on Lux. We could implement PIOS_RCVR_NumDevs() and if it returns >0 skip initing PPM. Should avoid issues with setup wizard/noise etc.

mlyle, 2016-05-18:
Or just have a flag whether to run PPM.  There's nothing wrong with having multiple RXes.  (Also, does GCS receiver count?)

tracernz, 2016-05-18:
Yah, that requires user action though -> support requests inbound. :-P

mlyle, 2016-05-18:
;) ppm on/off doesn't sound so bad, to fix the rare cases where PPM is a problem. 

IMO it's mostly useful for developers.  You can put a solder glob over the whole jumper, and switch between serial RXes and PPM without concern (vs. where serial to PPM will create an interrupt storm).

mlyle, 2016-09-01:
Ehhhh. No one's complainin'....

closed 2016-09-01 by mlyle