d-ronin/dRonin #1023: UAVOMSPBridge sends altitude with wrong sign on F3/F4

tracernz, 2016-06-02:
Sends PositionActual.Down as altitude, which is opposite direction to MSP protocol and what MWOSD expects. See ShikOfTheRa/scarab-osd#245

https://github.com/d-ronin/dRonin/blob/014d78244bc98f566394e4d8f56c1fe3ac202255/flight/Modules/UAVOMSPBridge/UAVOMSPBridge.c#L461

dustin, 2016-06-02:
By opposite, do you mean the sign is just flipped?

tracernz, 2016-06-02:
Yeah, AttitudeActual.Down is how far below our reference point we are, whereas MSP wants how far above.

tracernz, 2016-06-02:
GPS speed is also not right.

henn1001, 2016-06-03:
I knew i forgot something ...  never tested it myself as stated in #892 and then forgot it because of merge :/

cpusmith, 2016-06-03:
Once the speed is fixed I'll test it.

tracernz, 2016-06-03:
@henn1001 no prob, do you want to fix?

PR #1035 (2016-06-17): UAVOMSPBridge: Fix altitude and GPS speed scaling
Flip sign on altitude (MSP wants Up, PositionActual UAVO is NED), scale GPS speed correctly (MSP wants cm/s, GPSPosition has m/s). Not tested.

Fixes #1023
closed 2016-06-17 by mlyle