d-ronin/dRonin #1025: Support for easy race lap timer irc code transmission

emil-nasso, 2016-06-07:
We have just started using easy race lap timer at our club for some racing practice. It's a pretty cool project. We use a raspberry pi and some ir receivers. The unit turns into a wifi-hotspot and you can access the website on the pi to set up races and see laptimes. There is also an android app that gets the laptimes via the api and announces times using the phones built in text-to-speech. And it's all open source. They have their own protocol for the ir-transmitters and you can build your own transponders using an arduino. 

It would be really awesome if this protocol could be supported by droning when using the BrainFPV RE1.

This is their website: http://www.easyracelaptimer.com/
This is the code that goes onto the arduino transmitters: https://github.com/polyvision/EasyRaceLapTimer/blob/master/Arduino/TransmitterAttiny85v0.3/TransmitterAttiny85v0.3.ino

I was reading the code for this in dRonin and it looks pretty straight forward there. https://github.com/d-ronin/dRonin/pull/963/files

I'm not a c-developer though. I have considered giving this a try but i could probably use some help/mentoring. :) 

I'm not sure exactly how this would be implemented. The dRonin code seems to generate a packet that is transmitted. The easyracelaptimer system has delays in between blinks, of different lengths.

So, is there any interest in this? Is anyone with better knowledge of this type of development (probably everyone here :) ) than me give some input on this?

mluessi, 2016-06-07:
Unfortunately this won't be possible for RE1. The reason is that the hardware required for the supported protocols (I-Lap, Trackmate) is different. For those protocols, the pulses are much shorter than for easy race lap timer and the IR led intensity needs to be about 10x higher.

I had originally looked into adding support for OpenLap, which uses the same (or very similar) protocol as ERLT but they then announced that they wanted to move to a new version of the protocol due to limitations of the old ERLT one (I assume: limited ID numbers, mis-detections when multiple quads pass close to each other due to not enough "checksumming" and long packet duration). OpenLap was overall also more widely used than ERLT (based on the BrainFPV survey results). The new protocol was supposedly similar to I-Lap/Trackmate, but unfortunately they have since abandoned their plans as they think video frequency based  systems will take the place of IR based ones.

In RE1 the currently implemented protocols are implemented in hardware, i.e., the STM32 only generates the packet and everything else is done externally. Additional protocols could be added using a STM32 timer, but the on time of the IR LED is limited to about 120 microseconds, as longer on times could damage the IR transmitter hardware (FET, LED itself). So it won't work for the ERLT protocol, which needs 600us on times.

emil-nasso, 2016-06-07:
Thanks for the reply. Doesn't sound like something that is wanted or needed then.

I'll look into the other side of things instead, the lap time software. :)

closed 2016-06-07 by emil-nasso