d-ronin/dRonin #1038: BrainRE1 randomly fails to load proper configuration

glowtape, 2016-06-15:
* Seen in version (release version or githash): brainre_test1, nightly build 385
* On flight controller (if applicable): BrainRE1

Issue details:

When powering up the multicopter randomly, it ends up unflyable with bad motor rattling upon control input, or doesn't react to any input at all (despite successfully arming, i.e. spinning props). If this issue shows up, upon connecting the GCS, the StabilizationSettings appear all over the place (such as shown on the attached screenshot). Clicking "Reload board data" on the Stabilization page loads the proper values 99% of the times, and instantly renders the quad operable. This has been tried live under battery power, i.e. if the quad powers up with bad values, while GCS telemetry is connected, and causes said uncontrollable behavior and motor rattling, hitting reload on the Stabilization page, proper values will be loaded and the issue goes away (at least up until power cycling).

This isn't however only happening on battery power. Bad values being loaded also happens when the FC is only powered via USB (tried different cables). This issue doesn't happen 100% of the times, it's possible to power up the quad/FC fine occasionally.

The OSD settings are usually also affected, when this issue occurs, altho the reload button on the OSD page doesn't appear to do anything.

Interestingly enough, other user configured settings like ActuatorSettings, AttitudeSettings, ManualControlSettings and such, don't appear to be affected (or if, then it hasn't happened often enough to notice).

It is also very sporadically possible to load bad values once a while, should you keep clicking "reload board data" on the Stabilization page.

[image]

mluessi, 2016-06-15:
I have not been able to reproduce this on `next` and nightly build 385 (Windows). @glowtape can you post the settings (.uav) file that you are using? I assume this is either a hw defect on the particular RE1 @glowtape has or maybe some race condition in the telemetry stack. Sometimes when I click "Reload Board Data" the FC reboots, which makes me suspicious. Has anyone else noticed that?

glowtape, 2016-06-16:
I've just given it another try. Erased the settings partition, set up a bunch of things, rebooted and power cycled it a few times, everything worked fine. I've exported the settings into an .uav and reimported them, another few reboots, no issues.

Then I left it sitting unpowered a few minutes, connected via USB and it entered a boot loop. I left it running, after an estimated 10-15 loops it then booted up successfully. Points more towards a hardware issue.

Here's the settings that worked and then not for a while (radio's not set up, but it's not like it matters):
[settings.txt](https://github.com/d-ronin/dRonin/files/317704/settings.txt)

mluessi, 2016-06-16:
@glowtape thanks for investigating. It does sound like a hardware issue; likely a problem on the SPI bus when it first boots; maybe the FPGA takes longer to initialize for some reason. The RE1 @glowtape has is on the way back to me for analysis.

mluessi, 2016-06-23:
I have the RE1 @glowtape had here and flashed it to the current `next` and imported his settings. So far I have no been able to reproduce the problem; I have connected / disconnected RE1 dozens of time and also used the Reload Board Data button and all settings are intact. I tried with and without an X4R-SB receiver connected (with antennas very close to RE1).

@glowtape did you have any other peripherals (e.g. video transmitter) connected?

mluessi, 2016-06-23:
OK, I was able to reproduce it once by leaving RE1 unpowered for an extended period and then connecting it to USB. I suspect it has something to do with the power-up sequence on this particular unit.

glowtape, 2016-06-23:
Here's all that was on my quad, besides the RE1:

* MPB
* 4x Kiss24 with an 25V 270uF Rubicon ZL each
* X4R-SB powered via RE1
* TBS Unify Pro on MPB 5V
* PZ0420m on MPB 12V

The X4R was mounted in the front of the 250 frame with at least 5-6cm distance from the RE1, antennas away to the top.

Given that I could repro it on USB, only the RX could have influenced it. I disconnected per your request a while ago and it still happened. Also, given USB mode issues and that my og Brain works fine on the MPB, I can assume that that piece works correctly.

The RE1 was mounted on nylon standoffs, I've tried different USB cables on the computer.

mlyle, 2016-07-06:
Closing based on CRC work / no other repro.

closed 2016-07-06 by mlyle