d-ronin/dRonin #1055: Flight: MPU6000 bottom orientations appear to be incorrect

jihlein, 2016-06-24:
* Seen in version (release version or githash): Latest Next
* On flight controller (if applicable): Any with MPU6000 mounted upside dowm

Issue details: The inverted MPU6000 orientations don't work correctly.  The sign inversions appear incorrect, and possibly the axis assignments.  I corrected the PIOS_MPU60X0_BOTTOM_90DEG orientation by removing the 2 inversions, and then it works properly.  The rest of the "BOTTOM" orientations need review.  Actually all the assignments need reviewing, as there's some axis swapping going on too that I don't understand.

mlyle, 2016-06-24:
Surely at minimum you need one inversion to put something on the bottom-- doesn't Z accelerometer switch sign, and Z gyro at minimum?

jihlein, 2016-06-24:
Yes, of course.  Look at the 90 top orientation vs 90 bottom orientaion, you will see the z axis data is inverted from top to bottom orientation, and that is okay.  It was the x and y that were inverted incorrectly.  If you experiment with the bottom 90 degree orientation, you will see short term the ADI go one way, then long term the other, indicating the gyro and accels are out of phase.  I suspect there are others with the same issue, but I don't have time to go thru them all.  I plan to when I get back from vacation, or someone can take a look if they want to earlier.

Date: Thu, 23 Jun 2016 20:19:52 -0700

mlyle, 2016-06-24:
What I think I'd really like to see, in general (and it's a bit tough to do) is to remove the gyro/accel sensor orientation stuff, and to add them to the board orientation rotation.  No need to do redundant work when it can be a single rotation.

mlyle, 2016-06-24:
(But it's quite all right for us to focus on fixing the rotations first).

jihlein, 2016-07-21:
Corrected with #1058.  Closed.

closed 2016-07-21 by jihlein