d-ronin/dRonin #1077: Auto update on DTFc not working

jhitesma, 2016-07-01:
* Seen in version (release version or githash): 85af0a4a
* On flight controller (if applicable): DTFc

Issue details:
Tried to update my DTFc now that it's in next.  Connected the board and it was detected and offered to update.  But when I hit "Yes" on update nothing happened.

Tried it on my Sparky next and it appeared to work...but after the final reboot board wouldn't connect.  Then tried my Sparky2 and Brain both of which worked as expected no issues.  Plugged the Sparky back in and it was detected as running 85af0a4a and I was able to re-load configs and get it working.  

But DTFc still doesn't update.  Detects but doesn't do anything after being detected and saying "yes"

Screencast of me trying to update the DTFc (and then sparky) here: https://youtu.be/zC4PHrKJgZ0

@tracernz said a debug log would help, but I need guidance on how to generate one.  Will also try a newer set of artifacts tonight to see if it's still happening.

jhitesma, 2016-07-02:
Just tried the latest nightly 37b7bde1 and same thing, detects the board but after hitting "yes" to upgrade nothing happens.

I confirmed that I could flash 85af0a4 manually with no issues, but even after doing that  37b7bde1 still does nothing after I hit "yes" on the detection screen.

tracernz, 2016-07-02:
@jhitesma The debug gadget will give you a log of lots of events, mostly USB related ones at the moment. You will need to click Window > Edit Gadgets Mode on one of your workspaces (e.g. Firmware) and then Window > Split (or Split Side by Side), then select the Debug gadget on the empty pane to view it.

[image]

jhitesma, 2016-07-04:
Knew there was something obvious I was forgetting.

`20:18:11[DEBUG]Adding  5840 55292 1 "360028000E57335239383420+BL" "DTFc" "DTF UHF"
20:18:11[DEBUG]usbmonitor detection cycle complete.
20:18:12[DEBUG]"EnterDFU:64 bytes sent"
20:18:12[DEBUG]"StatusRequest:64 bytes sent"
20:18:12[DEBUG]"StatusRequest:64 bytes received"
20:18:12[DEBUG]"Status:0"
20:18:12[DEBUG]FINDDEVICES BEGIN
20:18:12[DEBUG]"FINDDEVICES SENDING CAPABILITIES REQUEST BUFFER_SIZE:64"
20:18:12[DEBUG]"FINDDEVICES CAPABILITIES REQUEST BYTES_SENT:64"
20:18:12[DEBUG]"FINDDEVICES CAPABILITIES ANSWER BYTES_RECEIVED:64"
20:18:12[DEBUG]"Device ID=55041"
20:18:12[DEBUG]"Device SizeOfCode=212892"
20:18:12[DEBUG]"Device SizeOfDesc=100"
20:18:12[DEBUG]"BL Version=131"
20:18:12[DEBUG]"FW CRC=3216671927"
20:18:12[DEBUG]"Partition 0 Size 212892"
20:18:12[DEBUG]"Partition 1 Size 100"
20:18:12[DEBUG]"Partition 2 Size 16384"
20:18:12[DEBUG]"Partition 3 Size 16384"
20:18:12[DEBUG]"Partition 4 Size 16384"
20:18:12[DEBUG]"Partition 5 Size 0"
20:18:12[DEBUG]"Partition 6 Size 0"
20:18:12[DEBUG]"Partition 7 Size 0"
20:18:12[DEBUG]"Partition 8 Size 0"
20:18:12[DEBUG]"Partition 9 Size 0"
20:18:12[DEBUG]"EnterDFU:64 bytes sent"
20:18:12[DEBUG]"Downloading description partition..."
20:18:12[DEBUG]"StartDownload 2 Last Packet Size 11 Bytes sent 64"
20:18:12[DEBUG]"StatusRequest:64 bytes sent"
20:18:12[DEBUG]"StatusRequest:64 bytes received"
20:18:12[DEBUG]"Status:0"
20:18:12[DEBUG]"Download operation completed-- 100 bytes"
20:18:15[DEBUG]"Connection to bootloader successful"
20:18:15[DEBUG]"File loaded successfully"
20:18:15[DEBUG]Padding
20:18:15[DEBUG]"Ready to flash firmware"`

And that's it.  No errors but it never actually tries to flash.

PR #1082 (2016-07-04): gcs: dtfc: mark target as upgradable
fixes #1077
closed 2016-07-04 by tracernz