d-ronin/dRonin #1101: Objects failing to save on Sparky

tracernz, 2016-07-10:
* Seen in version (release version or githash): e991fe9
* On flight controller (if applicable): Sparky

Issue details:
Possibly on other boards too. Pretty sure this is due to the CRC changes. We should consider reverting and possibly revisiting in the next cycle because this stuff is already proven to work well on all targets except RE1.

tracernz, 2016-07-10:
Note to self: check exactly what happens when read crc fails. We must not so something like reset to default settings as that will be unsafe in some cases, but instead panic (which should stop ALL threads and stop outputs).

mlyle, 2016-07-10:
I think panicking could be bad, would be hard to recover from.

tracernz, 2016-07-10:
Well it is a serious hardware failure ;-). We can't continue to run so unless we have a way to stop that without panicking... The recovery would be rescue into bootloader.

closed 2016-07-10 by tracernz

mluessi, 2016-07-10:
@tracernz are you positive that the CRC changes caused this? I would have preferred to find the actual reason why it was failing instead of just reverting the CRC work, which I feel is important for RE1.

mlyle, 2016-07-10:
@mluessi if RE1 can't work dependably without the CRC thing, that's a big problem.

I was skeptical about the CRC work for Samsara, and it broke F3 (proven by bisection).  It turns out my fix didn't completely fix it.

When the objects fail to save, the LEDs stopped blinking for @tracernz, which is consistent with a PIOS_Assert() fail in systemmod.

If you can work very quickly to stabilize it, we can consider it... but this has been part of why the release is hung up / late.

mluessi, 2016-07-10:
@mlyle it's not that RE1 can't work reliably without it; it only showed up on one board and I can no longer reproduce it, but I think it's in general good to have this check and would make me feel better if we had it for RE1.

How about ifdefing it, so we can only enable it on RE1 for this release and then figure out how to make it work for all targets in the next release?

tracernz, 2016-07-10:
It needs to be proven stable on all targets before it comes back in IMO. Maybe it could come in for re1 only if necessary. Not being able to store and/or retrieve settings is such a critical and frustrating issue that we can't afford to mess with it this late.

PR #1102 (2016-07-10): Revert CRC changes for Samsara
Can be considered for Quixote, but it's had too much negative impact so far and we're too late already.

Reverts #1061 and #1074

Fixes #1101