d-ronin/dRonin #1112: Gathering symbol info on OSX doesn't work.

mlyle, 2016-07-14:
#1110 attempted to gather symbol info on OSX.  It doesn't quite work.

On release builds, all of the shared library symbol dumps are 0 bytes.

```
ls -al `find  . | grep sym$` | less
...
-rw-r--r--@ 1 mlyle  staff  101446 Jul 13 21:11 ./dRonin-GCS/BECAA202898B301DA7C2E
EC418CAF5610/dRonin-GCS.sym
-rw-r--r--@ 1 mlyle  staff       0 Jul 13 21:11 ./libAeroQuad.dylib/libAeroQuad.dy
lib.sym
-rw-r--r--@ 1 mlyle  staff       0 Jul 13 21:11 ./libAggregation.1.0.0.dylib/libAg
gregation.1.0.0.dylib.sym
-rw-r--r--@ 1 mlyle  staff       0 Jul 13 21:11 ./libBrainFPV.dylib/libBrainFPV.dy
lib.sym
-rw-r--r--@ 1 mlyle  staff       0 Jul 13 21:11 ./libBrotronics.dylib/libBrotronic
s.dylib.sym
...
```

Tagged into Samsara for now.

tracernz, 2016-07-14:
Doesn't work when GCS_BUILD_CONF=release (the packaging script sets GCS_BUILD_CONF=debug when EXTRACT_SYMBOLS=YES).

mlyle, 2016-07-20:
Fixed by #1113

closed 2016-07-20 by mlyle