d-ronin/dRonin #1121: config conversion crash

dustin, 2016-07-16:
'NoneType' object has no attribute 'ENUMR_ChannelGroups' (/base/data/home/runtimes/python27/python27_lib/versions/third_party/webapp2-2.3/webapp2.py:1528)
Traceback (most recent call last):
  File "/base/data/home/runtimes/python27/python27_lib/versions/third_party/webapp2-2.3/webapp2.py", line 1511, in __call__
    rv = self.handle_exception(request, response, e)
  File "/base/data/home/runtimes/python27/python27_lib/versions/third_party/webapp2-2.3/webapp2.py", line 1505, in __call__
    rv = self.router.dispatch(request, response)
  File "/base/data/home/runtimes/python27/python27_lib/versions/third_party/webapp2-2.3/webapp2.py", line 1253, in default_dispatcher
    return route.handler_adapter(request, response)
  File "/base/data/home/runtimes/python27/python27_lib/versions/third_party/webapp2-2.3/webapp2.py", line 1077, in __call__
    return handler.dispatch()
  File "/base/data/home/runtimes/python27/python27_lib/versions/third_party/webapp2-2.3/webapp2.py", line 547, in dispatch
    return self.handle_exception(e, self.app.debug)
  File "/base/data/home/runtimes/python27/python27_lib/versions/third_party/webapp2-2.3/webapp2.py", line 545, in dispatch
    return method(*args, **kwargs)
  File "/base/data/home/apps/s~dronin-autotown/py:1.391825220857430657/convert.py", line 35, in post
    imported = LogFSImport(self.request.get('githash'), datafile)
  File "/base/data/home/apps/s~dronin-autotown/py:1.391825220857430657/dronin/logfs.py", line 109, in __init__
    uavo_defs.from_git_hash(githash)
  File "/base/data/home/apps/s~dronin-autotown/py:1.391825220857430657/dronin/uavo_collection.py", line 100, in from_git_hash
    self.from_tar_bytes(web_data)
  File "/base/data/home/apps/s~dronin-autotown/py:1.391825220857430657/dronin/uavo_collection.py", line 70, in from_tar_bytes
    self.from_tar_file(t)
  File "/base/data/home/apps/s~dronin-autotown/py:1.391825220857430657/dronin/uavo_collection.py", line 56, in from_tar_file
    u = uavo.make_class(self, f)
  File "/base/data/home/apps/s~dronin-autotown/py:1.391825220857430657/dronin/uavo.py", line 277, in make_class
    parent_options = getattr(parent_class, 'ENUMR_' + field_name)
AttributeError: 'NoneType' object has no attribute 'ENUMR_ChannelGroups'

mlyle, 2016-07-16:
Was fixed by #1119, hopefully, so it should be a question of deploying this on the web service.

dustin, 2016-07-16:
Updated the web services.

closed 2016-07-17 by tracernz