d-ronin/dRonin #1122: Conversion warning

dustin, 2016-07-16:
This wasn't an error to the user, but may indicate incomplete conversion.

Error on attempting to adapt ActuatorSettings (/base/data/home/apps/s~dronin-autotown/py:1.394239192895852089/convert.py:71)
Traceback (most recent call last):
  File "/base/data/home/apps/s~dronin-autotown/py:1.394239192895852089/convert.py", line 45, in post
    actuatorSettings = imported['UAVO_ActuatorSettings']
KeyError: 'UAVO_ActuatorSettings'

mlyle, 2016-07-16:
Do you have the hashes this happened with?

It's probably a duplicate of #848-- someone has a build from a pull request or their own unmerged branch, and we fall back to using the last release's UAVOs.  

Alternatively, if they've never saved actuatorsettings before upgrade, this will happen.

dustin, 2016-07-16:
No, logging is insufficient here.  :(

closed 2016-07-16 by mlyle

tracernz, 2016-07-17:
Should instrument the code with python's logging class so this will be better in future.