d-ronin/dRonin #1123: BrainFPV in-flight reset

ufoDziner, 2016-07-16:
* Seen in version (release version or githash): Build 455
* On flight controller (if applicable): BrainFPV

Issue details:
I and my son have experienced in-flight resets of the BrainFPV FC.  Initially I thought that it was a power issue to the FC, but it it happening randomly across 3 different quads.  Build 455 was used to see what the initial startup message is after an in-flight reset.  Screenshot attached.

[image]

ufoDziner, 2016-07-16:
BOOT: UNDEFINED appears to be the normal startup message as well.

dustin, 2016-07-16:
Just to be extra clear, are you able to see this in the UAVO as well?  I
don't know why it'd be wrong, but if the UAVO disagrees, then it's a
different bug.

ufoDziner, 2016-07-16:
Requested screenshots

[image]
[image]

ufoDziner, 2016-07-16:
@dustin Where would I check for a uavo?

dustin, 2016-07-17:
I can't quite remember which one it is (on my phone), but it's system
something and has a boot reason in it.

mlyle, 2016-07-17:
SystemSettings.RebootCause I think.  Doesn't need a flight -- just is the startup reason.

ufoDziner, 2016-07-17:
Normal startup screenshot.

[image]

mlyle, 2016-07-17:
Not awesome :/

fujin, 2016-07-17:
#1126 should make it so we can see the proper reason now yea? @ufoDziner if you could upgrade to nightly tomorrow :sunglasses:

mlyle, 2016-07-17:
fw_sparky2, static stack analysis (with my dodgy tools) on next:

```
>$ stabilizationTask                   456      312        7
```
vs tanto:
```
>$ stabilizationTask                   432      304        7
```

So stabilizationtask stack usage has increased by 24 bytes.

Other notable differences:  picoc task improved from 4536 to 4168.   Autotune task went from 864 to 872.     Loggingtask from 416 to 528 (!), ActuatorTask from 424 to 464.

ufoDziner, 2016-07-17:
Copied from #1125 because I think it belongs here as well.:

My son flew 5 packs and I flew 4 on two different quads using build 459 that includes this PR with no issues. Funny thing is that I broke mine after the 4th flight, so I took out the backup quad that was still on 455 without this PR. Halfway through the first pack it rebooted. So, I'm pretty confident that the issue is fixed.

ufoDziner, 2016-07-20:
Flew 2 more packs today, with no resets.

mlyle, 2016-07-20:
Believed to be resolved.

closed 2016-07-20 by mlyle

ufoDziner, 2016-07-31:
Had 1 in-flight reset, and 2 resets after a crash.  I don't believe that the crash was caused by the reset, but it's possible that they were.  The in-flight reset was with an OG brain using bfe4b825.  The 2 crash resets were with an RE1 was using 5f25d7d4.  "INDY WDOG" was displayed on the screen as the reboot reason each time.  These reboots happened somewhere mid-flight (I don't have an accurate time), but powered on for at least 5 min.