d-ronin/dRonin #1133: OSX: breakpad catches EXC_BREAKPOINT

tracernz, 2016-07-18:
This makes it incompatible with debugging generally, and causes it to crash the app immediately when starting a debug session through QtCreator.

mlyle, 2018-03-18:
https://github.com/google/breakpad/blob/67649c61853108eb0c29703f6ff0db42e9d69f10/src/client/mac/handler/exception_handler.cc#L120

tracernz, 2018-03-18:
So I want to just disable crash reporter when a debugger is connected. There is some good Apple sample code but I could not find what license it came under. Any ideas?

mlyle, 2018-03-18:
https://developer.apple.com/library/content/qa/qa1361/_index.html   is the sample code in question.

You can't really copyright "how to invoke an API" so it's OK to use that as a reference IMO.

mlyle, 2018-03-18:
Other sources:
https://books.google.com/books?id=K8vUkpOXhN4C&pg=PA734&lpg=PA734&dq=get+kp_proc+mac+os+x&source=bl&ots=OMjgSWZuYD&sig=FpyioyFpOxWumh55WvMDivCCo80&hl=en&sa=X&ved=0ahUKEwj9mPv-iPXZAhUIi1QKHVmfBLIQ6AEIUDAJ#v=onepage&q=get%20kp_proc%20mac%20os%20x&f=false

https://gist.github.com/s4y/1173880/9ea0ed9b8a55c23f10ecb67ce288e09f08d9d1e5

tracernz, 2018-03-18:
Yup, this is one I can quite easily handle if you like.

mlyle, 2018-03-18:
I love it when you handle things :D

PR #2131 (2018-03-19): GCS: Disable crash reporter when debugger connected
Currently only on macOS (where it's a problem), and Windows (not sure?). Harder to do on Linux and not a problem there. 😛 

Fixes #1133
closed 2018-03-19 by mlyle