d-ronin/dRonin #1136: Administrative: Samsara postmortem

mlyle, 2016-07-20:


tracernz, 2016-07-27:
* Really old bug bit us with the lack of CCM RAM BSS initialisation (#1139, #1140).
* Late build infrastructure changes caused grief.

mlyle, 2016-07-27:
* Stack overflow got us, and wasn't spotted by our test for a long time despite a lot of users running it / flying it for unknown reasons.  This is probably the one that distresses me most.

pug398, 2016-07-27:
Haven't had a flight issue on CC3D, Sparky1 or Revo. Probably not serious enough to grab peoples attention. With the CC3D after the legacy loader was flashed it couldn't be detected and a rescue then manual flash was needed. Sparky and Revo upgraded smoothly and have been flying without incident. Need to get Samsara.1 on them.

mlyle, 2016-07-27:
@pug398 Samsara.1 is identical to Samsara except a small change to BrainRE1 memory layout. 

The CC3D-upgrade-not-quite-reliable on Windows is annoying and needs further study for sure.

pug398, 2016-07-27:
CC3D has been problematic from the beginning and it ended up working fine. You guys are doing a great job and the platforms upgraded so far are flying smooth. Will get to Quanton and Sparky2 shortly.

pug398, 2016-07-30:
With my sparky2 and oplm I see the hwsparky2 uavo not being locked in sometimes and gcs does not recognize the hardware. At times it appears to "pass by" the uavo without stopping to pick it up. You can see the hardware page flash briefly when it is going to happen but then goes back to unknown board type page and hwsparky2 uavo has defaults.

pug398, 2016-08-13:
Not a big deal but spelling of screen is wrong in definition of OSD screen page settings. 
Was it decided to leave the model view orientation alone and have it track the PFD instrument rather than actual model orientation?

mlyle, 2016-08-13:
I believe model view tracks actual model orientation-- the previous default model is "backwards" and has the front pointing at what most would consider the back of the quad, but we changed that..

The red arms are the front of the quad is how I see it, and it looks right to me..

It's better if you open separate issues for these thigns-- the purpose of this issue is to roll up a set of things that went wrong with the samsara release and process so we can do better next time.  Actual issues we should address should go into real issues or they're likely to get lost.

pug398, 2016-08-13:
Ok so you did fix the model view after Samsara. Some of my quads have red arms in the front and I noticed the model view was still backwards. Sorry I am still somewhat behind times :)

mlyle, 2016-08-13:
It should be fixed *in* samsara.

pug398, 2016-08-13:
No it is backwards on all of mine. If you tip the quad nose up (red arms) the nose in the model view goes down (red arms). You can point the quad nose north and the heading will show north with the red arms in model view pointing north. The red arms in model view tip down when the quad nose is tipped up. Samsara.1

mlyle, 2016-08-14:
Yah, I'm sorry, I misspoke.  The green arms are in the front. In the initial view, the red arms face the user and the green arms are in front.   If the front of the flight controller faces the monitor when it is plugged in, they move together.

:( Can you please open new issues for each topic?

closed 2016-09-01 by mlyle