d-ronin/dRonin #1139: BrainRE1 does not init BSS stuff in CCM-SRAM properly!

mlyle, 2016-07-22:
This has triggered a bug on next with the new manualcontrol stuff, but---- probably triggers lots of undefined behavior elsewhere.

Putting it into Samsara.  We will have to hotfix.  Even though ill effects have not been observed on Samsara, tehy're surely there.

tracernz, 2016-07-22:
Ick.
Nobody zeros it:
https://github.com/d-ronin/dRonin/search?utf8=✓&q=_cmm_start

How normal BSS is zeroed: https://github.com/d-ronin/dRonin/blob/080150a5b271fcfb3fe36aa1e20f92fc0d956aca/flight/PiOS/Common/Libraries/ChibiOS/os/ports/GCC/ARMCMx/crt0.c#L326

mlyle, 2016-07-22:
For clarity, other targets have CCM and it actually comes up zero (though we should explicitly zero it to be "careful").

'446 doesn't have CCM, and we've moved some of the BSS away from the other stuff so it's not zeroed.. where it lands on other ram that has undefined contents on startup.

mlyle, 2016-07-22:
While this is still super bad, closing in favor of just #1140 based on further understanding.

mlyle, 2016-07-23:
reopened, because it is indeed bad.

PR #1141 (2016-07-23): RE1: Remove CCM section in Linker Definition
References #1140. confirmed @ 565a761
[image]
[image]
[image]

Edit by mpl: Fixes #1139
closed 2016-07-23 by mlyle